springfall2008/batpred #612: New feature - ability to save current settings to automation

springfall2008, 2024-01-13:
I'm experimenting with a feature that creates a file called '/homeassistant/predbat_restore_settings.yaml' which contains the current user configuration.

If you copy the contents of this into a new automation in HA it will, when run, restore your current HA settings for Predbat.

The idea is then users can save one or more sets of configuration and quickly change between them.

The alternative is to use a select drop down and have the user be able to save a new setting or restore an old one based on slots or suchlike.

Any feedback?

springfall2008, 2024-01-13:
I should have said the above is live in the current release but needs testing and documentation

PianSom, 2024-01-13:
Thanks for all the work - I am still getting to terms with the many, many options and trying to work out which ones work best for me. So a quick way to save/change settings would be fantastic.

I am a Home Assistant Core user, with Appdaemon in a different Docker container. Just as a note, the settings .yaml  is found in the appdeamon root folder (the one above "apps"). (I found the dashboard yaml there too - was wondering where that was!)

Personally, I'd have a preference for the drop-down alternative - saves me having to do the cli shuffle. But I'd be happy to manage the .yaml's with a cli and then have a drop-down populated from a dedicated profile folder if that was easier.

dochtie, 2024-01-14:
Thanks for this! Great idea. Could you also add a dropdown for “DEFAULT”? An easy way to quickly restore all settings back to the defaults.

Boffinboy, 2024-01-14:
I agree this would be quite useful, including the default option. Drop down would be preferable. Thank you for the continued improvements!

springfall2008, 2024-01-15:
I've changed this feature now (it's on main) to use a dropdown menu with save and restore options, please have a look and feedback.

PianSom, 2024-01-15:
Looks good so far. 

For docs - the backups are stored in a folder called "predbat_save" alongside the "apps" folder.

Is this change in 7.14.44 (and all points forward)?

mpartington, 2024-01-16:
Seems to work ok for me.  Maybe worth documenting that you can manually save the file name to something more meaningful using file editor (unless there's an easier method I've missed).

Is there a way to switch between the settings via a HA automation, or is it internal to Predbat?  My use case is currently I need a different set of settings for say a DFS session (e.g. turn off combine charge windows), so at the moment my external automation has a number of call service lines to update the relevant fields, then change them back. It would be simpler just to use the profiles. Not critical for me, as it's already set up and working, but maybe of benefit as a feature.

PianSom, 2024-01-16:
I am just back from holiday, so have `holidays_days_left` set. I see this is in the saved settings file.  I wonder if this is a setting that one might actually in practise want to save + restore?

springfall2008, 2024-01-16:
@mpartington by pop-ups do you mean notifications?

@PianSom actually its a good point, I probably shouldn't restore holiday_days_left , or the iboost status or the manual car soc level

mpartington, 2024-01-16:
Yes, notifications

closed 2024-01-20 by springfall2008

mpartington, 2024-01-20:
Seeing a bug I suspect is related to this. Everytime I restart HA, it seems to change my settings to a different saved configuration

springfall2008, 2024-01-20:
Do you have a log, it should be recorded?

mpartington, 2024-01-20:
Actually it doesn't appear to be all settings, but read only and min metric improvement definitely changed following restart at 18:46.  Attached log, but can turn debug on if required

[appdaemon1.log](https://github.com/springfall2008/batpred/files/13998935/appdaemon1.log)
[access.log](https://github.com/springfall2008/batpred/files/13998936/access.log)

mpartington, 2024-01-21:
As usual, this was nothing to do with Predbat...

Appears what is happening is the octopus savings session binary sensor is briefly switched on following a reboot. When it switches off, it triggers a local automation to put Predbat into a particular combination of settings.

gcoan, 2024-01-21:
Yes I found the same had happened, my Christmas lights were on at 1:30 am. Looked at the automation traces and found it was the binary sensor for saving sessions that had fired

gcoan, 2024-01-21:
@mpartington I have raised this as a bug on the Octopus integration https://github.com/BottlecapDave/HomeAssistant-OctopusEnergy/issues/716

mpartington, 2024-01-21:
Brilliant, thanks! As a workaround, I've added a helper that I switch on myself upon detection/signup of a saver session and turn it off at the end of a session as the last thing.  I've added a condition that that sensor must be true to restore a predbat config. Has the bonus I can now call up a config for the session (that will always be the same) and just revert to the ' previous.yaml' that has my latest preferences.

gcoan, 2024-01-30:
@mpartington Dave has released an update to the Octopus integration to fix this issue and it seems to work fine