springfall2008/batpred #614: Target reserve repeatedly set and then unset

dochtie, 2024-01-14:
**Describe the bug**
Predbat sets reserve and then a minute later resets it back to default. This is repeated each run. In the log it shows for example. 

2024-1-14 09:15:04.968266 INFO pred_bat: Adjust reserve to target charge 40 % (set_reserve_enable is true)

2024-01-14 09:15:04.975912 INFO pred_bat: Inverter 0
Current Reserve is 4.0 % and new target is 41 %

2024-01-14 09:15:15.511151 INFO pred_bat: Set inverter 0
reserve 41 via REST successful on retry 0

2024-1-14 09:16:27.770555 INFO pred_bat: Adjust reserve to
default as SOC 36% is above target 40 %

2024-01-14 09:16:27.780450 INFO pred_bat: Inverter 0
Current Reserve is 41.0 % and new target is 4.0 %

**Expected behavior**

Should not adjust reserve to default as the SOC is not above the target.

**Predbat version**

7.14.40

**log**

Log file attached
[appdaemon.log](https://github.com/springfall2008/batpred/files/13930935/appdaemon.log)

springfall2008, 2024-01-14:
I don't spot a specific issue here, in the 9:10 run Predbat used freeze charging to try to hold the SOC% at the current level, but during the period the SOC feel below the target and so it turned on charging and thus off the reserve setting. 

Are you seeing this happening quite a lot, I do have an idea that might resolve it in your system (which takes a while to calculate the plan) but re-reading the inverter settings between plan calculation and plan execution.

dochtie, 2024-01-14:
Okay. Thanks Trefor. I do see it a lot. I think the “ SOC 36% is above target 40 %” confused me since it’s not. Maybe just the wording. Appreciate you looking at this. My plan has been very unstable the last couple of days with agile rates being fairly flat.

dochtie, 2024-01-14:
@springfall2008 amazing to see an update already resolving this minor issue. Thank you. Incredible work.

dochtie, 2024-01-15:
Hi @springfall2008 It might be worth another look. My inverter is still having its reserve set and unset, even within 30 seconds. Here is a log snippet. I'm now on 7.14.42

2024-01-15 13:38:34.519761 INFO pred_bat: Reset inverter settings to safe mode (set_charge_window=True set_discharge_window=True force=)
2024-01-15 13:38:34.527533 INFO pred_bat: Inverter 0 Current Target SOC is 100 already at target
2024-01-15 13:38:34.535682 INFO pred_bat: Inverter 0 Current Reserve is 99.0 % and new target is 4.0 %
2024-01-15 13:38:45.260928 INFO pred_bat: Set inverter 0 reserve 4 via REST successful on retry 0
2024-01-15 13:38:45.322315 INFO pred_bat: Inverter 0 Adjust force discharge to False, change times from 00:00:00 - 00:00:00 to None - None
2024-01-15 13:38:45.328122 INFO pred_bat: Charge window will be: 2024-01-15 13:30:00+00:00 - 2024-01-15 15:30:00+00:00 - current soc 100 target 100
2024-01-15 13:38:45.334861 INFO pred_bat: Setting ECO mode as no discharge window planned
2024-01-15 13:38:45.342037 INFO pred_bat: Inverter 0 Adjust force discharge to False, change times from 00:00:00 - 00:00:00 to None - None
2024-01-15 13:38:45.349256 INFO pred_bat: Resetting charging SOC as we are not within the window or charge is disabled and inverter_soc_reset is enabled (now 01-15 13:35:00 target set_soc_minutes 30 charge start time 01-15 13:30:00)
2024-01-15 13:38:45.356996 INFO pred_bat: Inverter 0 Current Target SOC is 100 already at target
2024-01-15 13:38:45.364553 INFO pred_bat: Adjust reserve to target charge 100 % (set_reserve_enable is true)
2024-01-15 13:38:45.373490 INFO pred_bat: Inverter 0 Current Reserve is 4.0 % and new target is 99 %
2024-01-15 13:39:12.071793 INFO pred_bat: Set inverter 0 reserve 99 via REST successful on retry 1

dochtie, 2024-01-15:
[image]

dochtie, 2024-01-16:
[image]

dochtie, 2024-01-17:
im also seeing charge rate repeatedly changed sometimes within a few seconds. See attached. I can send a full log if preferred?
[image]