springfall2008/batpred #617: Home Battery Prediction not scaling for 2 batteries

Sycamore967, 2024-01-15:
**Describe the bug**
I have two 9.5kWh batteries.  Prebat is working out I have 19.1kWh hours available but the Home Battery Prediction graph only scales up to 9.5 kWh

**Expected behavior**
I'd expect the scale to show me the total capacity so I can see the top of the chart lines.

**Predbat version**

7.14.41

**Environment details**

- Inverter and battery setup - V2 inverter and two 9.5 kWh batteries
- Standard HAOS on a Pi
- I installed Predbat yesterday, so it's still bedding in.

**Screenshots**
I've included a full overview

**Log file**
Don't think you need it for this issue.
[image]

springfall2008, 2024-01-15:
You just need to edit the chart scale in the yaml

Sycamore967, 2024-01-15:
Ah brill - -sorry, still very new to HA and Predbat.  I looked but obviously missed it.

closed 2024-01-20 by springfall2008