springfall2008/batpred #619: predbat.plan_html exceed maximum size of 16384 bytes

Sycamore967, 2024-01-15:
**Describe the bug**
Log size is being exceeded - can this be configured?

**Expected behavior**
Plan file size should be big enough

**Predbat version**

7.14.41

**Environment details**

- Inverter and battery setup - V2 inverter - 2 batteries
- Standard HAOS installed on Pi
- Anything else? - New install, so it's probably me - I can see a solcast error in the log as well, I don't know if its related

**Screenshots**
N/A

**Log file**
Attached
[home-assistant.log](https://github.com/springfall2008/batpred/files/13938502/home-assistant.log)

springfall2008, 2024-01-15:
You can ignore this, as it’s expected, but in the documentation there is some information on how to disable this warning also

gcoan, 2024-01-15:
> You can ignore this, as it’s expected, but in the documentation there is some information on how to disable this warning also 

@Sycamore967 Have a look at the faq's, there's details there of how to suppress the message. It's just a warning and can be ignored

Sycamore967, 2024-01-15:
Brill thanks. Got it sorted. No errors now. Best regards,Mark

closed 2024-01-20 by springfall2008