springfall2008/batpred #620: Car Charging Slot carrying over into Peak time

dandwhelan, 2024-01-15:
**Describe the bug**

I am using the "Car Manual SOC" and "Car Charging Plan Smart" and use the smart slot binary switch to trigger my car charging. The predbat plan decided to keep the car charging in the peak Octopus Agile pricing between 4pm and 7pm when lots of other cheaper spots were available after in the day. One side note that may have contributed to this behaviour. The car charging window started during the day and a new automation I was trying turned it off straight away as it was charging during the day. (needs some tweaking)

[carcharger](https://github.com/springfall2008/batpred/assets/79152287/7f453ced-c2af-4273-b48d-5a93dedd8afb)

1- Car Charging Slot - Started via Predbat automation. Car Stop (my automation to stop during the day)
2- Car Charging Slot - Started via Predbat automation. Car Stop (my automation to stop during the day)
3- Car Charging Slot - Started via Predbat automation. Car Stop (my automation to stop during the day)
4- Started after I discovered the Car Stop was interfering. 
5 - Post 4pm, I manually turn the charger off as the slot had not fished and moved over to 4pm. 

[appdaemon.log](https://github.com/springfall2008/batpred/files/13941054/appdaemon.log)
[appdaemon1.log](https://github.com/springfall2008/batpred/files/13941056/appdaemon1.log)
[appsyaml.log](https://github.com/springfall2008/batpred/files/13941057/appsyaml.log)
[image]

**Expected behavior**

Charge the car in the cheapest windows.

**Predbat version**

v7.14.41

**Environment details**

8.2Kw Gen 2 Hybrid Inv

springfall2008, 2024-01-21:
So the issue here is that you are using 'car_charging_now' sensor which forces predbat to schedule the charge. Once this is set the only way for the car to stop is via your own automation.

If you are using Predbat slot to control the car then you should comment out 'car_charging_now' from apps.yaml.

It maybe I can fix this in future to avoid the issue, its a circular dependancy.

springfall2008, 2024-01-21:
Docs updated here: https://github.com/springfall2008/batpred/pull/642

dandwhelan, 2024-01-29:
Thanks alot. Works a treat!

closed 2024-01-29 by dandwhelan