springfall2008/batpred #626: Can we have a legend for the plan HTML?

jameslavery, 2024-01-17:
**Is your feature request related to a problem? Please describe.**
The plan HTML is very informative, but I'm not exactly clear as to what all the colours and symbols mean. I can guess but...

**Describe the solution you'd like**
Could we have documentation which gives a legend? This would be really useful. Unless I'm missing existing documentation!

**Describe alternatives you've considered**
I've looked at the documentation, but can't find it.

**Additional context**
N/A

jameslavery, 2024-01-17:
If I'm supplied with the information, I'd be happy to update the documentation!

gcoan, 2024-01-17:
I am working through updating and expanding the documentation for Predbat, haven’t done the full work on the output docs yet, but will do soon.

I’ll do this as part of that.

If you have any specific questions on the chart, please ask

gcoan, 2024-01-17:
The colours indicate how well the item compares to a baseline (which for the rates is a bit complex).

Basically green is good (cheaper import), yellow less good and red bad (high import)

The same for your house load (green = lower predicted load, etc)

Pv is the converse going grey red yellow green for increasing solar generation 

The arrows indicate the direction of travel so cost going up or down and battery soc going up, down or staying constant 

The states are all explained in this part of the documentation:

https://springfall2008.github.io/batpred/what-does-predbat-do/

I'll try to prioritise getting onto this

gcoan, 2024-01-18:
Phew, that was a lot longer to write the documentation for the plan card than I thought it would be.
It's now complete and I've included it in PR #625 but would welcome a second pair of eyes to see that it makes sense as there's a lot to it.
Draft of the new documentation is at https://github.com/gcoan/batpred/blob/main/docs/predbat-plan-card.md

jameslavery, 2024-01-18:
That's exactly what I was looking for, thank you.

Looks pretty spot-on. Minor point that Car and iBoost aren't shown if they're not configured (at least that's the case for my installation.

I've also got funky little suns (at least, that's what I assume they are) in my PV kWh column:

[image]

gcoan, 2024-01-18:
> Looks pretty spot-on. Minor point that Car and iBoost aren't shown if they're not configured (at least that's the case for my installation).

It should say that in the detailed explanation of each column that cars and iboost are only displayed if configured.  I have added it to the summary as well now.

> I've also got funky little suns (at least, that's what I assume they are) in my PV kWh column:

Yes, they are little sun symbols.

gcoan, 2024-01-20:
Completed as per #625 which has now been merged into main.

If you have any further feedback or comments on the documentation, please do re-open or open a new issue

closed 2024-01-20 by gcoan