springfall2008/batpred #629: Dashboard issue in 7.14.45 in monitor mode

dochtie, 2024-01-17:
**Describe the bug**
When set to monitor mode the dashboard in 7.14.45 shows an immediate charge to 100% and then all slots (except 1630) as hold charge 100%. Rolling back to 7.14.44 gives a correct dashboard. Updating again to 7.14.45 gives the incorrect dashboard.

**Expected behavior**
In monitor mode the dashboard shows the correct settings and plan in 7.14.44. On 7.14.45 is shows 100% charge and hold except for 16:30.

**Screenshots**
Screenshots of plans
[image]
[image]

**Log file**
Log file attached
[appdaemon 2.log](https://github.com/springfall2008/batpred/files/13968123/appdaemon.2.log)

springfall2008, 2024-01-20:
https://github.com/springfall2008/batpred/pull/636

springfall2008, 2024-01-20:
Released in: https://github.com/springfall2008/batpred/releases/tag/v7.15.0

closed 2024-01-20 by dochtie