springfall2008/batpred #630: Predbat set Eco mode not force discharge in the Energy Saving Session

gcoan, 2024-01-18:
**Describe the bug**
at 3pm Predbat's plan for the energy saving session was:
[image]

And at 5:30pm it executed that plan, and set the inverter to Eco mode to satisfy the predicted (historic) load.

The historic load (2.84/2.17kW) seems reasonable, in the peak rate there's usually cooking going on, the heat pump is on, and the Gen 1 hybrid inverter (max 2.6kW discharge rate) is supplying most of the house load with a bit of excess.

However as this was a saving session we had deferred the cooking and had turned the heat pump off, so the house load was far far lower than the historic normal.

To over-ride this behaviour I initiated a forced discharge, then set Predbat to read-only which of course cancelled the force discharge, so had to initiate another force discharge....

**Expected behavior**
With exceptionally high export rates (such as a saving session) I would expect Predbat to set a force discharge rather than Eco mode to ensure that as much energy as possible is exported.

**Predbat version**
7.14.45

**Environment details**
HAOS
Gen 1 hybrid, 5.2kW battery

**Screenshots**
At the start of the saving session, the plan was:

**Log file**
[appdaemon.log.4.txt](https://github.com/springfall2008/batpred/files/13973866/appdaemon.log.4.txt)

springfall2008, 2024-01-18:
Did you have metric keep set that prevented this?

gcoan, 2024-01-18:
> Did you have metric keep set that prevented this?

no best soc keep is 0.1, best soc min 0, metric battery cycle cost 0

I think the issue is that under normal peak rate circumstances, Eco mode would be absolutely fine, but not in a saving session.  Historic Home load is higher than the battery max discharge rate and by 5:30pm the predicted load was higher (due to load scaling of 1.15 - reflecting that if its a cold day and the heat pump has been on a lot, its likely to be on more)

springfall2008, 2024-01-18:
I think what is needed is and assumed load reduction during saving session

gcoan, 2024-01-18:
> I think what is needed is and assumed load reduction during saving session

Yes something like that because at the moment Predbat has no way of knowing that we have dramatically reduced our consumption.
Maybe the proposed enhancement to denote exceptional load (doing a big wash) that was suggested (can't find the issue number) might cover both?

JonathanLew1s, 2024-01-19:
I've always liked the idea of splitting the load, separating the base load from variable load. 
Having an entity, that is manually maintained, with your expected baseline 30 minute load. Using that base, rather than historic, as an optional for certain scenarios (holiday, saving sessions etc)
In my case that would be 0.195kWh per 30 minutes, the bare minimum to keep the fridges, freezers, servers and gadgets running that are always on. In vacation or saving session windows the variable loads, above 195w, are not planned.

springfall2008, 2024-01-20:
Released in: https://github.com/springfall2008/batpred/releases/tag/v7.15.0

gcoan, 2024-01-21:
The new input_number.predbat_load_scaling_saving should do what I want, I have set it to 0.1 to indicate I'm running at 10% of normal predicted load during the saving session.
This % is a guess, a specific kWh base load might be more accurate, but I'm expecting it'll still have the desired effect of forcing a discharge rather than going to Eco mode in a saving session.

closed 2024-01-21 by gcoan