springfall2008/batpred #633: Unexpected Discharge

JonathanLew1s, 2024-01-18:
**Describe the bug**
Both yesterday, and today, I've seen random discharge events on the current slot, not planned in advance. The export rate in the discharge slot is not lower than upcoming import slots and import is planned a few hours later at a higher rate.
Discharge started after the saving session yesterday, none of the further events align with a full SOC.

**Expected behavior**
Discharge when positive cost benefit within the window of the plan. 

**Predbat version**

7.14..45

**Environment details**

- Giv 5 Gen 2, 2x 9.5kWh

**Screenshots**
[image]
[image]

**Log file**
[_appdaemon_logs-10.txt](https://github.com/springfall2008/batpred/files/13979675/_appdaemon_logs-10.txt)

JonathanLew1s, 2024-01-19:
Again this morning. Discharging at 13.97 to charge at 14:56 in an hour.

[image]

JonathanLew1s, 2024-01-19:
The capacity would make sense to import if the demand deemed it necessary, but I cannot see any reason to discharge to only import later at a higher unit rate.

springfall2008, 2024-01-20:
Looking at this the first one is marginal, probably would be cured by setting metric_battery_cycle to 0.5p or 1p.
The second one is a discharge freeze, exporting solar (so no losses) and seems a little marginal again (the above would probably cure it also).

JonathanLew1s, 2024-01-20:
Understood, makes sense now you point out it's only exporting solar rather than a battery export. Agree the 1st discharge is less than 1p per unit so marginal.

Setting predbat_metric_min_improvement_discharge to 0.5 would achieve a similar outcome to setting a metric_battery_cycle? I like to see actual losses in the calculation/html rather than artificial inflating with the battery metric.

closed 2024-01-20 by JonathanLew1s