springfall2008/batpred #643: Suboptimal plan

slopemad, 2024-01-21:
It's been a while since I've used Predbat in anything other than read only mode, it's charge decisions have been bizarre at times.  This is tonight's plan, where it wants to hold the battery during the expensive part of the evening (say, 15-18p) (thus running off the grid), and charge up whilst prices are still comparatively high (say, 9-14p) and not charging when prices are at their lowest (say, 0p-8p), whilst to me it would be better to let the battery completely run flat this evening, and do a full recharge when prices are lowest (01:30-05:30).

It just feels the plan is more random than based on any kind of calculation, compared to how it used to be.

Logs at https://pastebin.com/U4RRBVVY

I have tweak second pass enabled, best soc keep at 0.5, Battery Cycle Cost set to 
[image]

springfall2008, 2024-01-21:
Can you try with metric battery cycle at 1.0 or less as I think your value of 3 is skewing towards grid use, it effectively adds 6p per kWh to the cost of using the battery

slopemad, 2024-01-21:
Well, that's looking more promising.  I guess I misunderstood how it used the metric battery cycle in the calculations, I guess it adds 3p to both charge and discharge?

[image]

springfall2008, 2024-01-21:
Yes correct, the explanation is in the manual but I don’t think it’s clear how this will influence grid use

slopemad, 2024-01-21:
That said, I've just noticed it's decided the car charge slot should be *now* rather than in the cheap slots :-|

springfall2008, 2024-01-21:
Do you have car charging plan smart on?

springfall2008, 2024-01-21:
If so check your ready by time too

slopemad, 2024-01-21:
Yes, because in the original plan at the top of the thread, the car charges were at the expected slot, but all I changed was the metric battery cost.  Ready Time is set to 07:00:00

springfall2008, 2024-01-21:
Can you attach a log, something is not right there?

slopemad, 2024-01-21:
OK, so car charging smart had somehow become off - I've turned it back on and it's fine now.
But the plans are now taking 4-5 minutes to create.
`now complete after 223.270022 seconds (limit=10)`
I have tweak optimisation on, full second pass is off and calculate regions is off too.

It's also doing a lot of charge/discharge in the same slot, which may well not work in reality if the heat pump kicks in, which it's likely to at those prices.  'Calculate charge on discharge' is off.

slopemad, 2024-01-21:
A log for the last run at 215 seconds... https://pastebin.com/nnfc9gWw

springfall2008, 2024-01-21:
If you haven’t already maybe try plan forecast hours as 24 to speed things up?

There can only be one charge and one discharge in a 30 minute period and they can’t overlap.

[image]

slopemad, 2024-01-22:
The issue was that, for some reason, a load of settings had changed, so smart car charge became disabled, it disabled expert mode (and thus only applied a bunch of defaults, which seem to make the run time substantially longer).  Working much better now.

springfall2008, 2024-01-22:
Odd, I’ve seen that myself once also but never tracked down why the settings changed

gcoan, 2024-01-25:
> Yes correct, the explanation is in the manual but I don’t think it’s clear how this will influence grid use

I've just completed documenting a new Predbat FAQ to better explain how the losses and the battery metric work.  Draft here: https://github.com/gcoan/batpred/blob/main/docs/faq.md#the-plan-doesnt-charge-or-discharge-when-i-expect-it-to

Also updated the main customisation text https://github.com/gcoan/batpred/blob/main/docs/customisation.md#battery-loss-options to highlight that the battery metric cost applies to both the charge and the discharge

closed 2024-02-10 by springfall2008