springfall2008/batpred #644: Battery not discharging when batpred thinks it is

timporter, 2024-01-21:
**Describe the bug**
Predbat shows the battery as 'discharging'. However, in the GivEnergy dashboard it shows the battery as idle and the house as running off the grid.

Read Only mode disabled, predbat mode "Control Charge & Discharge"

**Expected behavior**
During my peak window, it sometimes makes sense to discharge the battery to the grid and do a forced export, usually this works. On occasion predbat is saying it is discharging but when I look on the GivEnergy web UI or at the mobile app, I can see its actually importing from the grid and the battery is idle.

This is frustrating because these are usually peak price times.

The expected behaviour would be for the status given by the GivEnergy dashboard and the status given by predbat to match. I can see that during these times the battery % has not dropped, so it seems that it is the predbat status that is the wrong one and not the GE dashboard.

**Predbat version**

v7.14.40  A few minor versions behind.

**Environment details**

Inverter: GIV-HY3.6
Battery: 9.52kWh
Install through HACS

**Screenshots**

Chart from GE shows some grid imports and no battery usage for periods around 18:15-19:00

[image]

You'll have to take my word for it that the HA dashboard showed the status as "Discharging", as did the html plan (Discharge).

**Log file**
[predbat log.txt](https://github.com/springfall2008/batpred/files/14002542/predbat.log.txt)

timporter, 2024-01-21:
I should add that in general everything is fine, predbat can happily charge the battery during the off-peak window, decide when to run off the battery and when to run off the grid, etc, even sometimes the forced export works. Its not like it is completely unable to control things.

gcoan, 2024-01-21:
Sometimes I find the inverter just gets confused. It happened yesterday to me during the peak period, predbat set the inverter to Eco mode but it just didn’t start picking up the house load and so I was importing when predbat said idle.
I went into the GivEnergy portal and did a reset to defaults and that cured the problem for me.  Might be worth trying that

timporter, 2024-01-24:
Thanks, I tried that and a similar situation has happened this evening.

PredBat shows 'discharging'

[image]

The plan says 'discharge'

[image]

Yet it is importing from the grid (SmartMeter / IHD agrees) and battery is idle.

[image]

Any ideas?

Kind regards,

Tim.

gcoan, 2024-01-24:
Hi Tim,

There's two potential scenarios I can think of that are happening, there may be more, but if we can understand what's happening better in the battery it will help identify what the problem is.

Scenario 1.
The battery has been instructed to discharge but the house load is higher than the inverter discharge rate, so you are grid importing.
If for example you have the cooker on you could well be drawing more than the inverter max output.  Is it a gen 1, 2 or 3 hybrid?  The gen 1 is only 2.6kW.

Scenario 2.
Problem with the instruction not reaching the inverter or the inverter ignoring it for some reason.

My inclination is to scenario 1, but lets try to diagnose first.  One thing to bear in mind is that the data you see in the Givenergy portal is only sampled every 5 minutes so it may not be what is happening right now.  If you use the givenergy app and connect it in local mode, its every 30 seconds, but still not totally real time.

I'd suggest you add to your dashboard a power flow meter which gives a much better indication of what is going on.  Speaktothegeek has an excellent video on this, search for 'givenergy powerflow plus'.
Its available on HACS as a front end integration.

Secondly I would add a number of other inverter control sensors.  These you can use to see what the inverter mode is set to, and change it, set charge & discharge windows, etc.  If you've got Predbat running then Predbat will overwrite any changes you make next 5 minutes it runs, so if you want to change things you'll need to set Predbat to read only mode.  But the controls are useful to see what is happening with your inverter.

Here's an extract from my dashboard.  Add your inverter name (mine is prefixed h_, yours will be givtcp_)

          - type: horizontal-stack
            cards:
              - type: entities
                entities:
                  - entity: sensor.h_xxxx_soc
                  - entity: select.h_xxxx_mode
                  - entity: select.h_xxxx_force_charge
                  - entity: number.h_xxxx_target_soc
                  - entity: select.h_xxxx_charge_start_time_slot_1
                  - entity: select.h_xxxx_charge_end_time_slot_1
                  - entity: switch.h_xxxx_enable_charge_schedule
                  - entity: select.h_xxxx_temp_pause_charge
                  - entity: number.h_xxxx_battery_charge_rate
                  - entity: select.h_xxxx_discharge_start_time_slot_1
                  - entity: select.h_xxxx_discharge_end_time_slot_1
                  - entity: switch.h_xxxx_enable_discharge_schedule
                  - entity: number.h_xxxx_battery_discharge_rate
                  - entity: number.h_xxxx_battery_power_reserve
                title: G395 Inv2 Control
              - type: custom:power-flow-card-plus
                entities:
                  battery:
                    entity: sensor.h_xxxx_battery_power
                    state_of_charge: sensor.h_xxxx_soc
                  grid:
                    entity: sensor.h_xxxx_grid_power
                    invert_state: true
                    use_metadata: false
                  solar:
                    entity: sensor.h_xxxx_pv_power
                    display_zero_state: true
                  home:
                    subtract_individual: true
                  individual1:
                    entity: sensor.extension_power
                    name: Extension
                    icon: mdi:home-outline
                    color: red
                    display_zero_tolerance: 50
                  individual2:
                    entity: sensor.ashp_power
                    name: ASHP
                    icon: mdi:heat-pump-outline
                    color: '#9007DD'
                    display_zero_tolerance: 50
                clickable_entities: true
                display_zero_lines:
                  mode: show
                  transparency: 50
                  grey_color:
                    - 189
                    - 189
                    - 189
                use_new_flow_rate_model: true
                w_decimals: 0
                kw_decimals: 1
                min_flow_rate: 0.75
                max_flow_rate: 10
                max_expected_power: 2600
                min_expected_power: 0.01
                watt_threshold: 1000
                transparency_zero_lines: 0
                title: G395 PowerFlow+

Obviously delete the ASHP entry on the power flow, or add something else you want to monitor the house load going to.

It'll look like this when running:
[image]

This shows at the moment, no solar generation, 2.6kW coming off the grid and into the home, most of which is going to the heat pump, some to the extension - probably the kettle on.

This will be updated based on the polling frequency you have in GivTCP, so typically 10-30 seconds.

My guess is that even though the mode is set to Forced Export and there is a discharge slot active, the house load is exceeding the battery output so it has to grid import.

gcoan, 2024-01-25:
A more exciting power flow showing the battery charging, solar generation, the heat pump on. There's a negative figure on house consumption because I have 3 solar arrays and only 1 is shown on this powerflow.  The other 2 are generating 0.8kW at the moment but they're not configured on the powerflow
[image]

timporter, 2024-01-25:
Hi @gcoan 

Thank you for taking the time to write all of that up, I'll run some experiments over the weekend and get back with my results.

Kind regards,

Tim.

timporter, 2024-01-26:
I think I've identified the issue.

This evening it happened again, it was a nice sunny day so the battery was nearly fully charged at the start of the peak window. Batpred kept changing its mind about if it should be force exporting or running in eco mode, but thats by the by. At the times it wanted to be force exporting it was doing so, until towards the end of the peak window where this issue that I've been having came up again. Batpred said "discharging" but the battery was idle, so the house load was coming from the grid.

[image]

It jogged my memory that the times this has happened before have often been towards the end of the peak window.

Long story short, I can see in the GivEnergy portal historic graphs that during a period when I am exporting, if the battery passes below 25% it just stops and switches to import. eg...

From today, it switched to import as the battery goes below 25% despite there being at least half an hour of peak window left (ends at 19:00)
[image]

Earlier in the week, switches to import as the battery goes below 25% despite ~45 mins remaining of peak window
[image]

(Theres more of these as I go through the history).

So it got me thinking, maybe the issue is with GE and not batpred/HA.

I put batpred into read only mode and used the controls within GE dashboard to charge the battery up, then I set it to manually force export the battery, despite there being a good while left on my manually configured force export window it stopped when I hit below 25%.

[image]

Beyond that I can still run in eco mode, even with quite a high load and the battery happily discharges to provide that, but refuses to force export if I try and set that up again.

I think, for whatever reason, this is something GE is enforcing and then batpred is unaware its happening, which means the battery/inverter goes to take the house load from the grid and the battery sits idle. If batpred could be made aware this was going on it could command the system back to eco mode and that would be ok.

My system is quite new, last month, so maybe its something with newer versions of the battery / inverter unless you know of some hidden setting about this.

[image]

[image]

[image]

Thanks for taking the time with this!

I've either missed something, or this is some new behaviour with whatever versions of GE equipment I have because I can't see anyone else with the same issue.

Kind regards,

Tim.

springfall2008, 2024-01-26:
Hi Tim,

I think you have an issue with your setup I'm afraid, I suggest you contact GE and ask them to look into why your battery refuses to continue to discharge below 25% as it should go down to 4%

One thing to check is if you log into GE cloud, go to the remote control page, click the little circle arrow on each and every box and then snapshot the screen we can confirm your settings are okay.

gcoan, 2024-01-26:
Hi Tim, yes it does sound like something odd with the inverter.

I can see from the controls you've copied that predbat is doing a forced export down to 8% as the battery reserve level so it shouldn't just stop when it reaches 25%, it should continue down to the reserve predbat set.

You've done a "reset to defaults" which should clear anything weird on the inverter, but just go and look in the portal at inverter settings/ battery options- do you see anything there?
On my gen 1 hybrid I've got reserve 4% usable 96%, charge and discharge power. I have a recollection that there's an Emergency Power Supply reserve as well which could be what's tripping you up.
I can't remember if this is user configurable in the portal or you have to get GivEnergy support to do it.

If it's a newish system and is always stopping at 25% then it sounds like this.  Predbat has no visibility of this.

Good to see you got the power flow and controls working, will give you a better idea what's happening

timporter, 2024-01-26:
Thanks!

[image]

That'd do it then? Thats in the remote control panel. Weird if so, thats after a "reset to defaults". I'll play with it tomorrow and see if it actually lets me force export lower than 25%, will report back.

Kind regards and thank you,

Tim.

gcoan, 2024-01-27:
> Thanks!
> 
> 
> 
> [image]
> 
> 
> That'd do it then? Thats in the remote control panel. Weird if so, thats after a "reset to defaults". I'll play with it tomorrow and see if it actually lets me force export lower than 25%, will report back.
> 
Yes I rather think that's it.

On my gen 1 hybrid inverter I see the start and end time, both of which are set to 00:00, but that's it, no lower soc % limi5 whatever that is supposed to mean.

Try 4% the same as the normal reserve soc?  I'm pretty sure this is used for the emergency power supply. If it is, and you don't have one wired in to your inverter then there's no need to have it >4%. If you want to reserve something for the eps then you'll need to tell predbat what the higher reserve is.

Hope this is it!

Cheers Geoffrey

springfall2008, 2024-01-27:
Interesting so there’s a new control for this type of inverter that predbat can’t see. 

Is it available in GivTCP?

timporter, 2024-01-27:
[image]

If I look at the soc_1, it shows it was adjusted last night so I'm pretty sure its that.

[image]

Kind regards,

Tim

springfall2008, 2024-01-27:
Thanks, I think I can change the code to control this setting also. I'll probably just change it in REST for now, I assume that's what you are using?

timporter, 2024-01-27:
Based on the default config file https://github.com/springfall2008/batpred/blob/main/apps/predbat/config/apps.yaml  I have lines 69-71 still, so I assume yes, even though I didn't delete the section that follows it.

Kind regards,

Tim

springfall2008, 2024-01-27:
This didn't work in GivTCP, which is a bit of a pain.

gcoan, 2024-01-27:
The fact that there are 10 SoC entities in HA makes me wonder if those different SoC's are for different charge and discharge slots?  I think the newest inverters support up to 10 charge/discharge slots and the portal has multiple start/end times (my gen 1 has only 1 charge and 2 discharge slots).

timporter, 2024-01-28:
> The fact that there are 10 SoC entities in HA makes me wonder if those different SoC's are for different charge and discharge slots? I think the newest inverters support up to 10 charge/discharge slots and the portal has multiple start/end times (my gen 1 has only 1 charge and 2 discharge slots).

Yes, in the "Remote Control" page there are 10 slots for discharging times, each one with its own start & end times and Lower SOC % Limit. Same for charge, 10 of them and individual 'Upper SOC % limit'. Within the main GUI it only has the one set of start and end times, though curiously it does label those times "Slot 1" suggesting the existence of the further numbered slots but not actually showing them.

timporter, 2024-02-01:
Just to report that today the system was able to discharge below 25% as commanded by batpred

gcoan, 2024-02-01:
> Just to report that today the system was able to discharge below 25% as commanded by batpred

Glad you got it sorted.  Weird that that slot 1 Soc was set to 25%

closed 2024-08-06 by springfall2008