springfall2008/batpred #645: PredBat has never set a discharge slot

Sycamore967, 2024-01-22:
**Describe the bug**
PredBats plans have never set a discharge slot

**Expected behavior**
I'm expecting Predbat to discharge the battery if there is spare capacity every evening before the charge window or in saving sessions.
(I end most week days with 6-8 kWh spare)

I've never seen PredBat plan any sort of discharge window and end most days with 6-8 kWh spare. PB didn't discharge in the last saving session.

**Predbat version**

v7.15.1

**Environment details**

Hi Trefor, I'm now 7 days in to a new install and still in monitoring mode and have followed your video to configure for Octopus Intelligent and a fixed Outgoing rate.  I'm also signed up for Octoplus.  (Predbat successfully signed me up to the last saving session.)

The weather has been terrible, so PV generation has been negligible. only a few 100 kWh per day.  I've essentially been running off the batteries everyday.  I have 19.1 kWh.

**Screenshots**

I'm including todays battery chart and plan.

[image]

[image]

**Log file**
Attached
[appdaemon.log](https://github.com/springfall2008/batpred/files/14006731/appdaemon.log)

springfall2008, 2024-01-22:
Monitor mode just reports what you have set up in your system and won’t change anything.

you can try changing to control charge and discharge and set read only mode if you want to prevent the inverter from being controlled.

for best results you may need to set “calculate discharge during charge” as true and perhaps turn off “combine charge”

Sycamore967, 2024-01-22:
I understand that monitor mode will just be that and won't be making changes, but I took it to mean that it would be planning as though it is in control, so I expected it to do that.  I expected it to show me my settings give me what I expect, before turning on a control mode.  Is that not the case?

I'll try altering those two settings to see if it helps, before switching modes.

springfall2008, 2024-01-22:
Monitor mode will not plan as if in control, to do this you need to change to control charge and discharge. You can combine this with read-only mode if you want to prevent control of the inverter. Monitor mode is just that - it monitors what your system is doing so the plan is just what you have set yourself.

Sycamore967, 2024-01-22:
Ah right - ok thanks for the clarification.

Sycamore967, 2024-01-22:
Well, that's made a difference - I've now set Control charge and discharge and set read only mode.  I've not made the other two settings yet, as I'd like to stick with the settings in your video and not make multiple changes at once.  I have discharge slots in the  plan, so very happy.
It's a bit off the topic of my original post, but related to monitor mode.  In that mode Predbat also stopped the battery discharge when I plugged the car in - it is the option that I have set, so it worked perfectly.  Again I didn't expect that in monitor mode.  I'm hoping I still have that in read only mode?

springfall2008, 2024-01-22:
I’m not sure if that’s a bug or a feature, probably a bug.

in read only mode it won’t do this.

Sycamore967, 2024-01-22:
ok - thanks - I'll see how I go for a few days more before coming out of read only.

gcoan, 2024-01-22:
I've added to the documentation to highlight more clearly the Monitor mode and added guidance to set the control mode and read only first.

This be in the next documentation push I send through

gcoan, 2024-01-22:
@Sycamore967 here's the updated install guide:
https://github.com/gcoan/batpred/blob/main/docs/install.md#ready-to-light-the-touch-paper

and explanation of modes https://github.com/gcoan/batpred/blob/main/docs/customisation.md#predbat-mode

Sycamore967, 2024-01-23:
Brilliant thanks for the update.  I've read the update and it makes a lot more sense now.

closed 2024-01-23 by Sycamore967