springfall2008/batpred #647: Clarification of how days_previous works

gcoan, 2024-01-22:
**Describe the bug**
The predbat documentation is inconsistent in how it describes days_previous in apps.yaml works https://springfall2008.github.io/batpred/apps-yaml/

From looking at the code I believe day=0 refers to today so days_previous of 1 is yesterday, and days_previous of 7 is the same day as today a week ago?

I *believe* that in a prior issue you advised not to use days_previous of 1 because that includes yesterday and today, which in part is resulting in ambiguity in the documentation where it describes averaging a complete week's consumption.

**Expected behavior**
Since clear definition of how days_previous works and some sample scenarios in the documentation that illustrate its use.

I am happy to update the documentation once we're clear on how it works.

springfall2008, 2024-01-22:
If you think about it then it’s consistent, 1 is 24 hours ago but if your prediction runs for more than 24 hours then it would wrap?

gcoan, 2024-01-22:
I've never really thought about the plan length (prediction duration) and days previous as being inter-twined.  

I know Predbat shows the historic load in the log, but had assumed this was a 24 hour sum of load (nothing to do with plan length).  And the historic load for a slot was calculated from the appropriate days_previous load.

e.g. if days_previous is 7 and 14 (so same day a week ago and a week prior) then

load in (say) 10am slot = days_previous_weighting of average(10-10:30am load from day 7, 10-10:30am from day 14)

Appreciate its not necessarily as simple as that as there's modal removal, scaling the load by factors, etc, but as a simple model, does this work?

If we are agreed that day 0=today, 1=yesterday, ... 7=same day a week ago

Then the example of averaging load over the prior week is wrong. It currently has days_previous as 2 through to 8.  It should be 1 through 7

Is there any issue of including day 1 in the days_previous?  If you take my approach of working out slot-wise load then I can't see there is an issue.
Trying to follow the logic if Predbat works load_history based on plan_hours, and plan_hours is say 36, and days_previous of 1 whether that will cause a problem or not.

springfall2008, 2024-01-22:
So effectively the load prediction for a time is 

Load_at(Future time - days_previous*24 hours ) * weight 

if future time is more than 24 hours ahead then you still end up in the future and so another 24 hours is subtracted. 

So 1 day ago is special in this regard, where as days previous of 2 or more isn’t

gcoan, 2024-01-22:
Thanks Trefor

Checking I understand,  with an example (and ignoring the weighting):

Time now Today is Monday, 10pm.

Days previous = 2
Load at for 9am tomorrow (Tuesday) will be:
09:00 tomorrow - 2 * 24 == 09:00 today - 1 * 24 == 09:00 yesterday Sunday

Days previous = 7
Load at for 9am tomorrow (Tuesday) will be:
09:00 tomorrow - 7 * 24 == 09:00 today - 6 * 24 == 09:00 Tuesday last week

Is this right?

This isn't how I expected it to work because its counting back from the future time, not the current time.

But OK if that's how it works, I can write this up in the docs.

This works fine for a forecast less than 24 hours ahead.

Continuing the example above, days_previous is set to 7, forecast for 36 hours ahead, i.e. Wednesday 10am is:

Load at Wednesday 10:00 this week - 7 * 24 == Wednesday last week 10:00 

Which is as I'd expect it to be.

If days_previous = 1
Load at for 9am tomorrow (Tuesday) will be:
09:00 tomorrow - 1 * 24 == 09:00 today

But if forecasting 36 hours ahead,
Load at for 10am Wednesday will be:
10:00 Wednesday - 1 * 24 = 10:00 Tuesday which is in the future, so a further 24 hours is subtracted, to arrive at load at(10:00 Today Monday)

If I got all that right, then the examples of days_previous in apps-yaml.md are actually correct, its not all that obvious why they work - will need more explanation

springfall2008, 2024-01-22:
Yes this all sounds correct to me.

gcoan, 2024-01-22:
Thanks, I will work out how to write this up in a less complex way !

closed 2024-01-22 by gcoan

gcoan, 2024-01-26:
Updated apps.yaml documentation to explain how this works
https://github.com/gcoan/batpred/blob/main/docs/apps-yaml.md#basics

@springfall2008 appreciate if you can check it makes sense