springfall2008/batpred #648: Battery Discharge Hold during Octopus Intelligent  - Pre planning phase

Sycamore967, 2024-01-23:
**Is your feature request related to a problem? Please describe.**
When I first plug my car in, it can take up to 30 mins for Octopus to stop the charge and create a plan. Predbat doesn't detect this and allows the battery to discharge to the car.  From the documentation I understand that the car_charging_from_battery switch also relies on Predbat detecting the Octopus Intelligent that's been set up.  This won't be set in the first charge phase or Pre-Planning phase as I'm describing it here.

**Describe the solution you'd like**
Use the car charging sensor already configured in apps.yaml to detect the charge power increasing and activate the car charging from battery block based on that during the Pre-Planning phase of Octopus intelligent.

**Describe alternatives you've considered**
Setting a scheduled charge window on the charger itself to prevent the initial Pre-planning charge.  Octopus still detect that the car is at home and plugged in and creates the Intelligent plan.  The downside of this is that the charger schedule either needs to be wide enough to cover any 'early' intelligent slots that may be created, or the schedule needs to be disabled just prior to the first charge window.  In either case it's not the flexible charge option that both Octopus and I am looking for.  Predbat could be the solution.

**Additional context**
I've noticed that once the Intelligent Schedule is created Predbat does its thing really well and blocks the battery discharge, but I've only needed to charge a few times outside of the normal 23:30-05:30 window.  Each time my trick of creating a charger schedule was in place for the initial 'plug in'. Tonight I made sure I turned the charger schedule off before connecting and observed the Predbat behaviour.

springfall2008, 2024-01-23:
So the way I managed my system is my car charger (Wallbox) has it's schedule set to 23:30pm until 5:30am. Then I have an automation in HA which detects Predbat slot being enabled and it enables the Wallbox to start charging if its not already, thus bypassing the schedule. A second automation stops the Wallbox if the predbat slot turns off again. 

Here's the video for this one: https://youtu.be/805Blbfc9Os

You could also link up 'car_charging_now' to a sensor which indicates if your car is charging or not, this will cause Predbat to stop the battery draining but it will also assume its a low rate slot.

Sycamore967, 2024-01-25:
Morning, ok I gave this ago last night
# In some cases car planning is difficult (e.g. Ohme with Intelligent doesn't report slots)
  # The car charging now can be set to a sensor to indicate the car is charging and to plan
  # for it to charge during this 30 minute slot
  car_charging_now: 're:(switch.hypervolt_charging)'
  #  - off

  # Positive responses for car_charging_now
  car_charging_now_response:
    - 'yes'
    - 'on'
    - 'true'

I updated car_charging_now: 're:(switch.hypervolt_charging)' but haven't changed anything else as you can see.

It didn't work - PredBat didn't stop the battery discharging when I plugged in and waited for Octopus to detect it.
I don't know if I've not updated the yaml correctly and I should have altered something else as well, or that I should have restarted HA - which I didn't do.

Can you let me know if I've updated the setting properly ?

Should I have used a sensor rather than a switch for instance?

springfall2008, 2024-01-25:
Did you check the logs and see if the switch.hypervolt_charging was linked up correctly and was detected? I assume you see this switch turn on in HA when you plugged the car in?

Sycamore967, 2024-01-26:
Hi, I've attached the log and settings files from last nights attempt.  I've change the Hypervolt sensor to one that detects 'Ready' or 'Charging'.  I didn't make any difference to the results but is a sensor rather than a switch.  I also updated the conditions to include 'charging' in a latter attempt - What is the # - off setting doing ?  Is this actually turning off the check so really is the source of my problem and should I have updated the response settings with 'charging' ?
[Latest settings.zip](https://github.com/springfall2008/batpred/files/14066419/Latest.settings.zip)

  car_charging_now: 're:(sensor.hypervolt_charging_readiness)'
  #  - off

  # Positive responses for car_charging_now
  car_charging_now_response:
    - 'yes'
    - 'on'
    - 'true'
    - 'charging'

At any rate, batpred still didn't stop the battery discharging when I plugged the car in.

springfall2008, 2024-02-01:
You may want to toggle 'octopus_intelligent_ignore_unplugged' to False also

springfall2008, 2024-09-01:
Closing this as its already doable with an automation in HA

closed 2024-09-01 by springfall2008