springfall2008/batpred #661: Settings disappeared or reset

iainfogg, 2024-01-29:
**Describe the bug**

I ran a few updates today, HACS plus a few HACS packages (none related to Predbat), then did a restart.

After it I noticed Predbat kept noticing me off status changes, even though I have that turned off.

Looking further, I noticed the mode had been set to Monitor, the Set Discharge During Charge was turned on, Expert mode was turned off, and I don't know about other settings.

I'm on 7.15.1 and have been for at least a few days I think.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Predbat version**

7.15.1

**Environment details**

- Givenergy 5.0 inverter, 2 x 9.5 batteries
- Standard HAOS installer

**Log file**

I've not had time yet to look through the logs to find the first entry after the restart but can do that later if needed.

springfall2008, 2024-01-30:
Hopefully I fixed this in the latest version, but we will have to wait and see

iainfogg, 2024-01-31:
Thanks. Should the new version also bring back the settings that I had, or just stop them from disappearing again @springfall2008 ?

springfall2008, 2024-01-31:
Only stop them disappearing again I’m afraid

closed 2024-02-03 by springfall2008