springfall2008/batpred #666: battery charge curve not being created by predbat

gcoan, 2024-01-29:
**Describe the bug**
Upgraded to 7.15.4 (after the crashes of 7.15.3 - thanks for the quick bugfix)

But the power curve is not created by predbat

getting the error
pred_bat: Note: Can not find battery charge curve, one of the required settings for soc_kw, battery_power and charge_rate are missing from apps.yaml

Looking at apps.yaml, none of these are set because I am using givtcp_rest and so all the inverter controls are commented out, vis:

```
  # When set use the REST API rather than HA entity for control, should be more reliable/faster to control
  # Set one per inverter
  # If using Docker then change homeassistant.local to the Docker IP address
  givtcp_rest:
  # - 'http://homeassistant.local:6345'
   - 'http://homeassistant.local:6346'

  # If not using REST then instead set the Control here (one for each inverter)
  # - you can delete this section if using REST
  #charge_rate:
  #  - number.givtcp_{geserial}_battery_charge_rate
  #  - number.givtcp2_{geserial2}_battery_charge_rate
  #discharge_rate:
  #  - number.givtcp_{geserial}_battery_discharge_rate
  #  - number.givtcp2_{geserial2}_battery_discharge_rate
  #battery_power:
  #  - number.givtcp_{geserial}_battery_power
  #  - number.givtcp2_{geserial2}_battery_power
  #pv_power:
  #  - number.givtcp_{geserial}_pv_power
  #  - number.givtcp2_{geserial2}_pv_power
  #load_power:
  #  - number.givtcp_{geserial}_load_power
  #  - number.givtcp2_{geserial2}_load_power
  #soc_kw:
  #  - sensor.givtcp_{geserial}_soc_kwh
  #  - sensor.givtcp2_{geserial2}_soc_kwh
 ```

**Expected behavior**
Battery curve created per new feature in 7.15.3/4

**Predbat version**
7.15.4

**Environment details**
HAOS
Gen 1 hybrid

springfall2008, 2024-01-30:
Unfortunately REST doesn’t provide any history so the fix is just to comment these back in.  I probably need to change the documentation to say that

gcoan, 2024-01-30:
OK, I didn't appreciate that, yes to update the documentation to make it clearer

gcoan, 2024-01-30:
Its not working for me either.

I uncommented out the three non-rest lines the charge curve needs:
```
charge_rate:
    - number.h_{geserial}_battery_charge_rate
  battery_power:
    - number.h_{geserial}_battery_power
  soc_kw:
    - sensor.h_{geserial}_soc_kwh
 ```

```
2024-01-30 10:28:40.623482 INFO pred_bat: Find charge curve with sensors sensor.h_sd2237g395_soc_kwh and number.h_sd2237g395_battery_charge_rate and predbat.status and sensor.h_sd2237g395_battery_power
2024-01-30 10:28:55.283062 INFO pred_bat: Find charge curve has 9.0 days of data, max days 9
2024-01-30 10:28:55.300701 INFO pred_bat: Note: Can not find battery charge curve, one of the required settings for predbat_status, soc_kw, battery_power and charge_rate do not have history, check apps.yaml
```
This is a slightly different error message from when I didn't have the sensors uncommented 

Note: Can not find battery charge curve, one of the required settings for soc_kw, battery_power and charge_rate are missing from apps.yaml

And I have a 5.2 battery with 80% DoD that reports as 5.22.  I have battery_scaling set to 0.8, assume that will be taken account of?

springfall2008, 2024-01-30:
It's probably a bug related to the battery scaling, I'll try to make a fix

springfall2008, 2024-01-30:
Please can you re-test on MAIN?

PianSom, 2024-01-30:
Looks good, Trefor. 

I commented the REST setting s and uncommented the others. For my AIO I got the following - 
2024-01-30 18:26:24.122216 INFO pred_bat: Charge curve can be entered into apps.yaml:
  battery_charge_power_curve:
    100 : 0.19
    99 : 0.19
    98 : 0.19
    97 : 0.45
    96 : 0.57
    95 : 0.57
    94 : 0.7
    93 : 0.82
    92 : 0.95

So I will now go and do exactly that.

springfall2008, 2024-01-30:
I’d comment that REST back in, you don’t need to disable it you just needed the sensors to be set.

battery rate max scaling is the charge rate scaling

I think you are confusing this with battery scaling ?

PianSom, 2024-01-30:
Yes, I reverted when I put the charge curve in, thanks.

And yes, about battery scaling. What should I be having as 0.85? Should it be the UI element called "Metric Battery Value Scaling" (which I think is input_number.predbat_battery_value_scaling?

(Someone who was  stupid could get confused around here! :) )

gcoan, 2024-01-30:
I tried the version on main, it looks as if it tried but still didn't work, sorry

```
2024-01-30 19:22:29.911807 INFO pred_bat: Find charge curve with sensors sensor.h_sd2237g395_soc_kwh and number.h_sd2237g395_battery_charge_rate and predbat.status and sensor.h_sd2237g395_battery_power
2024-01-30 19:22:57.583016 INFO pred_bat: Find charge curve has 9.0 days of data, max days 9
2024-01-30 19:22:57.591039 INFO pred_bat: Note: Can not find battery charge curve, one of the required settings for predbat_status, soc_kw, battery_power and charge_rate do not have history, check apps.yaml
2024-01-30 19:22:57.595890 INFO pred_bat: Found 1 inverters totals: min reserve 0.17 current reserve 0.21 soc_max 4.18 soc 0.17 charge rate 2.6 kW discharge rate 2.6 kW battery_rate_min 0.0 w ac limit 5.0 export limit 5.0 kW loss charge 5 % loss discharge 5 % inverter loss 3 %
2024-01-30 19:22:57.687054 INFO pred_bat: Base charge    window [  ]
```

[appdaemon (16).log](https://github.com/springfall2008/batpred/files/14103281/appdaemon.16.log)

my apps.yaml:
```

pred_bat:
  module: predbat
  class: PredBat

  # Import rates can be overridden with rate_import_override
  # Export rates can be overridden with rate_export_override
  # Use the same format as above, but a date can be included if it just applies for a set day (e.g. Octopus power ups)
  # This will override even the Octopus plugin rates if enabled
  #
  rates_import_override:
    -  date: '2024-01-25'
       start: '21:00:00'
       end: '23:30:00'
       rate: 0
    -  date: '2024-01-28'
       start: '10:00:00'
       end: '13:00:00'
       rate: 0
       
  # Days previous is the number of days back to find historical load data
  # Recommended is 7 to capture day of the week but 1 can also be used
  # if you have more history you could use 7 and 14 (in a list) but the standard data in HA only lasts 10 days
  days_previous:
    - 2
    - 3
    - 4
    - 5
    - 6
    - 7
    - 8

  # Days previous weight can be used to control the weighting of the previous load points, the values are multiplied by their
  # weights and then divided through by the total weight. E.g. if you used 1 and 0.5 then the first value would have 2/3rd of the weight and the second 1/3rd
  # Include one value for each days_previous value, each weighting on a separate line.
  # If any days_previous's that are not given a weighting they will assume a default weighting of 1.
  days_previous_weight:
    - 1
  
    # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # XXX: This is a configuration template, delete this line once you edit your configuration
  # template: True

  # Set to auto-match with a GivEnergy serial number, but you can override the serial or the sensor names
  # if it doesn't work or if you have more than one inverter you will need to list both
  geserial: 're:sensor.h_(.+)_soc_kwh'
  # geserial2: 're:sensor.givtcp2_(.+)_soc_kwh'

  # Sets the maximum period of zero load before the gap is filled, default 30 minutes
  # To disable set it to 1440
  load_filter_threshold: 30
[... 300 lines cut ...]
  # Creates an entity called 'binary_sensor.predbat_export_trigger_<name>' which will be turned On when the condition is valid
  # connect this to your automation to start whatever you want to trigger
  #export_triggers:
  #   - name: 'large'
  #     minutes: 60
  #     energy: 1.0
  #   - name: 'small'
  #     minutes: 15
  #     energy: 0.25

  # If you have a sensor that gives the energy consumed by your solar diverter then add it here
  # this will make the predictions more accurate. It should be an incrementing sensor, it can reset at midnight or not
  # It's assumed to be in Kwh but scaling can be applied if need be
  #iboost_energy_today: 'sensor.xxxxx'
  #iboost_energy_scaling: 1.0

  # Nordpool market energy rates
  futurerate_url: 'https://www.nordpoolgroup.com/api/marketdata/page/325?currency=GBP'
  futurerate_adjust_import: True
  futurerate_adjust_export: False
  futurerate_peak_start: "16:00:00"
  futurerate_peak_end: "19:00:00"
  futurerate_peak_premium_import: 14
  futurerate_peak_premium_export: 6.5
```

PianSom, 2024-01-30:
Geoffrey - I commented out my equivalent of your
```
  givtcp_rest:
  # - 'http://homeassistant.local:6345'
   - 'http://homeassistant.local:6346'
```
and then uncommented ALL of the section below. That seemed to do the trick.

(I then reverted to the original versions.)

gcoan, 2024-01-30:
Thanks @PianSom 
Its strange that the log file shows that the settings I made  to uncomment out the specific non-rest sensors that the charge curve needs are being picked up, and it looks to try to get the curve data.
As much as anything else would like to get to the minimum set of instructions we have to give to Predbat users to get the charge curve created.  So hence why I only uncommented the 3 specific lines (and I had to change them to match my givtcp setup, so trying to avoid doing this x lots!)

springfall2008, 2024-01-30:
> Yes, I reverted when I put the charge curve in, thanks.
>
> And yes, about battery scaling. What should I be having as 0.85? Should it be the UI element called "Metric Battery Value Scaling" (which I [...]

No battery_scaling is in apps.yaml and that controls the reported battery size, e.g. 0.85 would scale your battery size down by 15%

springfall2008, 2024-01-30:
@gcoan I don't really understand your setup, but can you charge the four entities it mentions in the logfile in the charge curve section and show a period where the battery goes from 90% to 100%?

gcoan, 2024-01-30:
@springfall2008 the apps.yaml is a bit of a mess because I have two inverters but at the moment only 1 has a working battery, so I commented out inverter 1 leaving inverter 2 now as the only inverter that predbat knows about and manages

my givtcp prefixes for the inverters are g_ and h_ (rather than givtcp and givtcp2_)

I believe I have apps.yaml set correctly, but happy to be corrected !

The first message appears to show that predbat has at least found all the required entities for the charge curve:
2024-01-30 19:22:29.911807 INFO pred_bat: Find charge curve with sensors sensor.h_sd2237g395_soc_kwh and number.h_sd2237g395_battery_charge_rate and predbat.status and sensor.h_sd2237g395_battery_power

The error message is:
2024-01-30 19:22:57.591039 INFO pred_bat: Note: Can not find battery charge curve, one of the required settings for predbat_status, soc_kw, battery_power and charge_rate do not have history, check apps.yaml

In apps.yaml, skipping all the commented out lines for inverter 1 which isn't configured at the moment in predbat:

```
  charge_rate:
    - number.h_{geserial}_battery_charge_rate
  battery_power:
    - number.h_{geserial}_battery_power
  soc_kw:
    - sensor.h_{geserial}_soc_kwh
```

and predbat.status is of course already defined in HA

Here's the 4 sensors from 12:50 to 13:15 today when the battery was being charged to 100%
[image]
[image]

springfall2008, 2024-01-30:
Interesting, I think I know what the issue might be - I suspect not all %'s are represented in your battery size? Maybe I could allow some gaps and fill them...

springfall2008, 2024-01-30:
That said you don't seem to have a curve anyhow!

springfall2008, 2024-01-30:
@gcoan please try the updated version on MAIN which might fix it or at least give more debug info if not

gcoan, 2024-01-30:
@springfall2008 I upgraded to main but the error looked the same
```
2024-01-30 21:22:18.230267 INFO pred_bat: Find charge curve with sensors sensor.h_sd2237g395_soc_kwh and number.h_sd2237g395_battery_charge_rate and predbat.status and sensor.h_sd2237g395_battery_power
2024-01-30 21:22:36.250591 INFO pred_bat: Find charge curve has 9.0 days of data, max days 9
2024-01-30 21:22:36.324883 INFO pred_bat: Note: Can not find battery charge curve, one of the required settings for predbat_status, soc_kw, battery_power and charge_rate do not have history, check apps.yaml
```

so turned debug on and then restarted appdaemon 

TBH the log looks the same to me?
[appdaemon (17).log](https://github.com/springfall2008/batpred/files/14104554/appdaemon.17.log)

gcoan, 2024-01-30:
> That said you don't seem to have a curve anyhow!

yea that's weird, because if I look at these sensors for a time period 2 days ago and re-do the screen shots that I put on #650 there is a charge curve!

[image]
[image]

springfall2008, 2024-01-31:
@gcoan can you re-run with an extra debug line:

                data_point = 99
                for minute in range(0, min_len):
                    **if (minute % 60) == 0:
                        self.log("Minute {} soc {}% soc_kwh {} soc_max {} status {} charge_rate {} battery_power {}".format(minute, soc_percent.get(minute, 0), soc_kwh.get(minute, 0),self.soc_max, predbat_status[minute], charge_rate[minute], battery_power[minute]))**
                    if soc_percent.get(minute, 0) == data_point and predbat_status[minute] == "Charging" and charge_rate[minute] == max_power and battery_power[minute] <= 0:

The bit in bold added inside the for loop?

gcoan, 2024-01-31:
Done, took a while to do as I've got a stinking cold, courtesy of Mrs C

Log starts from 5965
[appdaemon (18).log](https://github.com/springfall2008/batpred/files/14111934/appdaemon.18.log)

gcoan, 2024-01-31:
Just in case it was a factor, configured apps.yaml to comment out REST and set all the givtcp controller entities.
Charge curve still didn't work
[appdaemon (19).log](https://github.com/springfall2008/batpred/files/14114365/appdaemon.19.log)

PianSom, 2024-01-31:
Another user (The Black Cat) just said the exact same thing over on the forum. Very strange! I wonder why it worked for me.

springfall2008, 2024-01-31:
Just taken a look, I wonder if you never get 99% charge level?

gcoan, 2024-01-31:
I've just looked at a couple of my SoC graphs and they go 96, 98, 98, 100

PianSom, 2024-01-31:
FWIW I hit 100% pretty much every day. Only missed out once (98%) in the 10+ days I looked at the charts for

gcoan, 2024-01-31:
> Okay so I need to change the code to allow 99% to be missed

I wonder if the ‘skip step’ needs to be more generic. My battery appears to skip 99% but other batteries have different step profiles I think ?

gcoan, 2024-01-31:
The new release was auto-applied, but sorry to say that its crashing:

```yaml
2024-01-31 20:45:01.924900 INFO pred_bat: Find charge curve with sensors sensor.h_sd2237g395_soc_kwh and number.h_sd2237g395_battery_charge_rate and predbat.status and sensor.h_sd2237g395_battery_power
2024-01-31 20:45:03.665440 INFO pred_bat: Find charge curve has 9.0 days of data, max days 9
2024-01-31 20:45:03.678841 INFO pred_bat: Charge Curve Percent: 97-97 at 01-31 06:48:00 took 1 minutes charged 0.04 curve 0.0 average_power 2.425
2024-01-31 20:45:03.680718 INFO pred_bat: Charge Curve Percent: 96-96 at 01-31 14:42:00 took 1 minutes charged 0.04 curve 0.0 average_power 2.425
2024-01-31 20:45:03.686935 INFO pred_bat: Charge Curve Percent: 95-94 at 01-31 14:41:00 took 1 minutes charged 0.08 curve 0.0 average_power 2.425
2024-01-31 20:45:03.690974 INFO pred_bat: Charge Curve Percent: 92-92 at 01-31 14:38:00 took 1 minutes charged 0.04 curve 0.0 average_power 2.423
2024-01-31 20:45:03.694120 INFO pred_bat: Charge Curve Percent: 91-91 at 01-31 14:37:00 took 2 minutes charged 0.04 curve 0.0 average_power 2.422
2024-01-31 20:45:03.695377 INFO pred_bat: Charge Curve Percent: 90-90 at 01-31 14:35:00 took 1 minutes charged 0.04 curve 0.0 average_power 2.423
2024-01-31 20:45:03.696578 INFO pred_bat: Charge Curve Percent: 89-88 at 01-31 14:34:00 took 2 minutes charged 0.08 curve 0.0 average_power 2.423
2024-01-31 20:45:03.700465 INFO pred_bat: Charge Curve Percent: 86-86 at 01-31 06:36:00 took 2 minutes charged 0.04 curve 0.0 average_power 2.423
2024-01-31 20:45:03.701618 INFO pred_bat: Charge curve before adjustment is: {97: 0.0, 98: 0.0, 99: 0.0, 100: 0.0, 96: 0.0, 94: 0.0, 95: 0.0, 92: 0.0, 93: 0.0, 91: 0.0, 90: 0.0, 88: 0.0, 89: 0.0, 86: 0.0, 87: 0.0}
2024-01-31 20:45:03.702413 INFO pred_bat: ERROR: Exception raised float division by zero
```

[appdaemon (20).log](https://github.com/springfall2008/batpred/files/14117168/appdaemon.20.log)

I'll revert to the prior release

PianSom, 2024-01-31:
Eek - a predbat crash in the middle of a Power-up. Up there in "worst nightmare"-land :)

gcoan, 2024-01-31:
Hmm, bit of a flaw in the update processing.  Don't know if you can easily fix it (I can raise a new issue if you want), but as 7.15.9 is crashing I selected to update with 7.15.8, it updated, and then straight away auto-updated itself back to 7.15.9

Ideally would want auto-update to be paused or turned off if you manually select a specific version

Going to have to turn off auto-update
[Uploading appdaemon (21).log…]()

gcoan, 2024-01-31:
> Eek - a predbat crash in the middle of a Power-up. Up there in "worst nightmare"-land :)

No its fine, the battery is fully charging itself.
I do have problems with the power up events as the ones that are >2 hours predbat plans discharging then charging in the session to maximise the export revenue, but we have washing machine, tumble dryer etc on so are pulling more load than normal and I don't really want the battery to be unnecessarily discharging then charging again under these circumstances so have to create manual charge slots to force it to stop.  Not the end of the world and the extra discharges also not the end of the world

springfall2008, 2024-01-31:
New video added https://www.youtube.com/watch?v=L2vY_Vj6pQg

@gcoan just comment in the battery curve in apps.yaml and set it to 100: 1.0 

There is something wrong with your average power data as it's reporting 2.4 rather than 2400 - is the sensor in the wrong units?

springfall2008, 2024-01-31:
>so are pulling more load than normal 

You might like the new override feature, you can put in a load scaling also e.g:

```yaml
  rates_import_override:
   -  date: '2024-01-23'
      start: '11:00:00'
      end: '11:30:00'
      rate: 0
      load_scaling: 1.5
```

gcoan, 2024-01-31:
> @gcoan just comment in the battery curve in apps.yaml and set it to 100: 1.0
> 
> There is something wrong with your average power data as it's reporting 2.4 rather than 2400 - is the sensor in the wrong units?

I do have the majority of my power sensors in kW rather than watts.  Is Predbat not reading the unit of measure maybe?

The load_scaling, definitely useful feature, thanks

springfall2008, 2024-01-31:
> I do have the majority of my power sensors in kW rather than watts. Is Predbat not reading the unit of measure maybe?

Yes sorry its just reading it in watts as it is meant to be the GivTCP sensor

gcoan, 2024-01-31:
I've set the power settings to kW so its easier to read on dashboards etc
[image]

PianSom, 2024-01-31:
Bit cheeky to go off-topic I know, but while we are talking units it'd be VERY useful if the units were shown on the HA dashboard for inputs. I can never recall what is %, £ or kW (or possibly W) or something else

gcoan, 2024-01-31:
> I've put a divide by 0 protection on the version on MAIN.
> 
> As a future feature I think I could check the units, but right now all the units are assumed. I'm suprised that changed the data from HA!

I'm loading it now.  Yes changing the UoM seems to set what HA holds the entity as. I had the same problem with the GivTCP battery card that wasn't reading UoM either and kept thinking I was charging at 2.4W !!

... update

it seems that the charge curve is now working, but has come up with a bit of a weird one:

```yaml
2024-01-31 21:17:18.164374 INFO pred_bat: Find charge curve with sensors sensor.h_sd2237g395_soc_kwh and number.h_sd2237g395_battery_charge_rate and predbat.status and sensor.h_sd2237g395_battery_power
2024-01-31 21:17:20.056245 INFO pred_bat: Find charge curve has 9.0 days of data, max days 9
2024-01-31 21:17:20.068317 INFO pred_bat: Charge Curve Percent: 97-97 at 01-31 14:41:00 took 1 minutes charged 0.04 curve 0.0 average_power 2.425
2024-01-31 21:17:20.070273 INFO pred_bat: Charge Curve Percent: 96-96 at 01-31 14:40:00 took 2 minutes charged 0.04 curve 0.0 average_power 2.425
2024-01-31 21:17:20.071857 INFO pred_bat: Charge Curve Percent: 95-94 at 01-31 14:38:00 took 1 minutes charged 0.08 curve 0.0 average_power 2.425
2024-01-31 21:17:20.077867 INFO pred_bat: Charge Curve Percent: 92-92 at 01-31 14:36:00 took 2 minutes charged 0.04 curve 0.0 average_power 2.422
2024-01-31 21:17:20.081471 INFO pred_bat: Charge Curve Percent: 91-91 at 01-31 14:34:00 took 1 minutes charged 0.04 curve 0.0 average_power 2.424
2024-01-31 21:17:20.083107 INFO pred_bat: Charge Curve Percent: 90-90 at 01-31 14:33:00 took 2 minutes charged 0.04 curve 0.0 average_power 2.423
2024-01-31 21:17:20.084825 INFO pred_bat: Charge Curve Percent: 89-88 at 01-31 14:31:00 took 1 minutes charged 0.08 curve 0.0 average_power 2.424
2024-01-31 21:17:20.090611 INFO pred_bat: Charge Curve Percent: 86-86 at 01-31 13:58:00 took 1 minutes charged 0.04 curve 0.0 average_power 2.424
2024-01-31 21:17:20.092496 INFO pred_bat: Charge curve before adjustment is: {97: 0.0, 98: 0.0, 99: 0.0, 100: 0.0, 96: 0.0, 94: 0.0, 95: 0.0, 92: 0.0, 93: 0.0, 91: 0.0, 90: 0.0, 88: 0.0, 89: 0.0, 86: 0.0, 87: 0.0}
2024-01-31 21:17:20.095351 INFO pred_bat: Charge curve can be entered into apps.yaml:
  battery_charge_power_curve:
    100 : 0.0
    99 : 0.0
    98 : 0.0
    97 : 0.0
    96 : 0.0
    95 : 0.0
    94 : 0.0
    93 : 0.0
    92 : 0.0
    91 : 0.0
    90 : 0.0
    89 : 0.0
    88 : 0.0
    87 : 0.0
    86 : 0.0

2024-01-31 21:17:20.098406 INFO pred_bat: Consider setting in HA: input_number.battery_max_rate_scaling: 0.0
```

is this a side effect of the uom being kW?

gcoan, 2024-01-31:
I changed the battery_power sensor from kW back to watts, and forced an update to apps.yaml to reload predbat and the curve is still the same

springfall2008, 2024-01-31:
Yeh if you change the units back to watts it should work

gcoan, 2024-01-31:
> Yeh if you change the units back to watts it should work

which sensor?

springfall2008, 2024-01-31:
Battery power, your log file still shows 2.4 and not 2400

gcoan, 2024-01-31:
that's what I changed.  The extract of the logfile above was before I changed battery power.  I'll change it again and see what happens

gcoan, 2024-01-31:
Here's the logfile
[appdaemon (22).log](https://github.com/springfall2008/batpred/files/14117528/appdaemon.22.log)

Its still showing it as 2.4

but the sensor is changed:
[image]

so maybe the history is stored with the uom, which would be logical

springfall2008, 2024-01-31:
I’ll have a go at pulling the units when I get time

gcoan, 2024-01-31:
It’s not a top priority for me right now. Just as long as it doesn’t crash.Could always put in the charge curve as 100% then could go back to the main code stream if you don’t get to thisThanks

mpartington, 2024-02-01:
Just a query as keen to give a go now I've done a few full power charges. However, Not all charges are equal, I has a soc drop, so it spent ages faffing around to recalibrate the top limit next charge, which would throw the charge curve off. Since then I've done a few more typical charges, but what will predbat base the calculations on. Is it most recent data, or does it try and do some averaging across multiple runs?

springfall2008, 2024-02-01:
There's a new version on MAIN that averages the data points across all charges, also accounts for units. Finally you can set the charge curve in apps.yaml to 'auto' and it will just use the computed one.

gcoan, 2024-02-01:
I tried the new version @springfall2008 and it seems to work:

```yaml
2024-02-01 11:22:59.488771 INFO pred_bat: Charge curve before adjustment is: {97: 25.350000000000005, 98: 0.98, 99: 0.98, 100: 0.98, 96: 29.140000000000008, 94: 31.000000000000007, 95: 31.000000000000007, 93: 0.98, 92: 29.40000000000001, 91: 31.36000000000001, 90: 30.38000000000001, 88: 29.66000000000001, 89: 29.66000000000001, 86: 29.40000000000001, 87: 0.98}
2024-02-01 11:22:59.490987 INFO pred_bat: Charge curve can be entered into apps.yaml or set to auto:
  battery_charge_power_curve:
    97 : 0.89
    96 : 0.93
    95 : 0.99
    94 : 0.99
    89 : 0.98
    88 : 0.98
```

This was with predbat configured to use all the direct givtcp controls.

I then changed apps.yaml to put it back into REST mode, but just left the battery_charge_rate, battery_power and soc_kwh set to the givtcp sensors (for the history) to see if this 'minimal config' worked, and it did, but it gave a different curve without the 88 or 89% datapoints:

```yaml
2024-02-01 11:47:36.752957 INFO pred_bat: Charge curve before adjustment is: {97: 25.770000000000007, 98: 0.98, 99: 0.98, 100: 0.98, 96: 29.28000000000001, 94: 31.33000000000001, 95: 31.33000000000001, 93: 0.98, 92: 29.40000000000001, 91: 31.36000000000001, 90: 31.36000000000001, 88: 30.38000000000001, 89: 30.38000000000001, 86: 29.40000000000001, 87: 0.98}
2024-02-01 11:47:36.758043 INFO pred_bat: Charge curve can be entered into apps.yaml or set to auto:
  battery_charge_power_curve:
    97 : 0.91
    96 : 0.93
    95 : 0.97
    94 : 0.97
```

And I notice that neither have the full range of %'s that I see on my soc graph, including 100%!

>I've just looked at a couple of my SoC graphs and on two charges to 100% today they went 90, 91, 92, 93, 95, 96, 97, 98, 98, 100

springfall2008, 2024-02-01:
This is mostly as the values that are 1.0 aren't output, it seems your 98->100 is quicker than 97-98!

springfall2008, 2024-02-01:
Looks good

yeh the scaling is for both, I maybe should split out discharge scaling, I don’t know if both are the same or not

mpartington, 2024-02-01:
I'm fairly sure they are not the same, so would need splitting. The Giv
guys have confirmed a few times that the battery power is displayed in DC.
So my 3kW Inverter only displays 2880W ish on charge, but say 3150 W on
discharge

I've just forced it now

mpartington, 2024-02-01:
Mis
[image]
sed screenshot

springfall2008, 2024-02-01:
Ill try to split them out soon :)

gcoan, 2024-02-01:
> This is mostly as the values that are 1.0 aren't output, it seems your 98->100 is quicker than 97-98!

I've just updated to the 7.15.10 release and it's produced a different curve, this time with 99 and 100

  battery_charge_power_curve:
    100 : 0.46
    99 : 0.46
    98 : 0.46
    97 : 0.88
    96 : 0.93
    95 : 0.97
    94 : 0.97
    89 : 0.98
    88 : 0.98

springfall2008, 2024-02-01:
Maybe more sensible, now it’s averaging all data points

dochtie, 2024-02-01:
Great features here. I changed to auto successfully but got a wildly different curve from the previous version.

Version 7.15.9 seems a lot more realistic.

  #  91 : 1.0
  #  92 : 0.95
  #  93 : 0.82
  #  94 : 0.7
  #  95 : 0.57
  #  96 : 0.44
  #  97 : 0.32
  #  98 : 0.19
  #  99 : 0.19
  #  100 : 0.19
  
Version 7.15.10

INFO pred_bat: Find charge curve has 22.0 days of data, max days 22
 Charge automatically computed as:
  battery_charge_power_curve:
    100 : 0.22
    99 : 0.22
    98 : 0.25
    97 : 0.35
    96 : 0.54
    95 : 0.6
    94 : 0.68
    93 : 0.75
    92 : 0.76
    91 : 0.78
    90 : 0.78
    89 : 0.78
    87 : 0.92
    86 : 0.97
    85 : 0.68

mpartington, 2024-02-01:
An additional question to above... It you switch back to slow charge, will that artificially reduce the higher power (lower soc) points in auto,? As the charge rate is likely set below the point the BMS would limit

springfall2008, 2024-02-01:
Auto isn’t going to work in low power mode as eventually there will be no data for charging at full rate

gcoan, 2024-02-01:
If you set auto creation of the power curve does it cache it between runs as its quite resource intensive to create it from the history

[15 more replies]

closed 2024-02-03 by springfall2008