springfall2008/batpred #667: Flip flopping between charging and hold charge at 100% creating battery cycling

wilsondw, 2024-01-30:
**Describe the bug**
Since the changes made to address Freeze Charging being too keen, Predbat has typically planned to charge to 100% in overnight flux cheap rate and then hold charge for several hours at standard rate before allowing discharge later in the day. The plan is ok - albeit can cause some export when there is more sun that expected. However, in practice, instead of following this plan predbat keeps swapping between hold charging and charging which seems to cause the battery to oscillate between 98 and 100% charge.

**Expected behavior**
Would expect predbat to stay in either charge or hold charge rather than flip flopping between them. I suspect that the oscillation is not doing the battery any good.

**Predbat version**

7.15.2

**Environment details**

- Gen 2 with 9.5 battery
- Standard HAOS install

**Screenshots**
[image]
[image]

**Log file**
[appdaemon-4.1.log](https://github.com/springfall2008/batpred/files/14094432/appdaemon-4.1.log)

springfall2008, 2024-01-30:
I've released a fix, basically your max reserve setting is 99% and so the battery keeps dropping down to 99% and then being charged again. I have changed the behaviour so that if the current battery level is above the max reserve setting it will keep charging enabled and not use hold mode.

wilsondw, 2024-01-31:
Many thanks for the super fast response. Have downloaded the update 👍. Will monitor for a couple of days and let you know how it goes.

closed 2024-02-10 by springfall2008