springfall2008/batpred #673: Predbat set the inverter to charge in the peak period

gcoan, 2024-01-30:
**Describe the bug**
Predbat has been very reliable and consistent in choosing logical battery activities.

Ordinarily predbat will hold the battery charge in the start of the peak period for the first couple of slots, depending on rates, and then release it to meet house demand at 17:00.  The battery (only a 5.2 so 4.2 usable) then will last me through the worst of the peak rate slots.

But today at 17:35 (in the peak rate period) I got an alert that the battery was being set to charging.

Looking in predbat, this was correct, the predbat state was Charging and there was a charge slot set for 17:30-18:00.

The battery at the time was idle as the soc was close to the target charge of 64%.

So I left it.

Then when I checked at 17:50 I could see that the battery was now on 60% SoC and was charging at full rate.  The rates didn't make any reason for Predbat to charge at this time.

I set the 17:30 slot to Force idle and the Predbat status went to 'hold charging target 0%' which doesn't make any sense.

The battery however was discharging to meet house load.

**Expected behavior**
No charging in peak rate period when there's no cost reason to do so.

Hold charging at 0% doesn't occur.

**Predbat version**
17.5.5

**Environment details**
HAOS
Gen 1 hybrid, 5.2 battery

**Log file**
[appdaemon (15).log](https://github.com/springfall2008/batpred/files/14103091/appdaemon.15.log)

Current settings:
[24_01_30_19_10_00.txt](https://github.com/springfall2008/batpred/files/14103137/24_01_30_19_10_00.txt)

springfall2008, 2024-01-30:
The only odd thing I can see is combine_charge_slots maybe best set to Off, but I don't see why it would matter. Annoyingly I cleaned up the debug output a bit and now I don't have enough from the log to help that much. If it happens again can you turn on debug enable?

springfall2008, 2024-01-30:
Try turning combine charge off also.

gcoan, 2024-01-30:
Yes can turn combine charge off, but the recommendation for agile is to have it set on?

springfall2008, 2024-01-30:
Hmmm it shouldn’t really matter but in that specific case I think you might have had a 1 hour slot and the charging was in the second half

gcoan, 2024-01-30:
still don't know why it would want to charge in the peak rate.

Tomorrow's peak period is typical of what Predbat normally does, the battery is at 100% by 16:00, the 4.2kW SoC of my 5.2 is released in the 16:30, 17:30 and 18:00 slots
[image]

This is with combine charge slots off

The fuller plan with combine charge slots off
[image]

and with combine charge slots back on, a bit more discharge at 11:30 and no partial slot discharge at 13:00. but financially much the same result and the peak is the same
[image]

I'll leave combine charge slots off

gcoan, 2024-02-02:
I'll close this and reopen if it ever happens again (and I notice it!)

closed 2024-02-02 by gcoan