springfall2008/batpred #674: Change Predbat save settings file extension from .yaml to .txt

gcoan, 2024-01-30:
**Is your feature request related to a problem? Please describe.**
I think it would be useful if we added instruction to save a copy of the current predbat settings and attach those to new issues.  This would make it easier to debug predbat issues caused by incorrect config settings.

At the moment the predbat settings save file is a .yaml file - is there a reason for this?

If its a .yaml file then you can't attach it to github (at least in the Windows browser you can't):
[image]

If I rename it in HA to .txt and then download it as a txt, then of course it can be attached to github

**Describe the solution you'd like**
Predbat settings save files are saved as .txt to make it easier to attach them to github

**Describe alternatives you've considered**
Continue with .yaml files but have to add instructions to rename them in HA before downloading to Github - likely to put inexperienced users off

**Additional context**
Of course if there is a reason for .yaml or another file extension that could be chosen, that is more compatible with github file upload

springfall2008, 2026-08-24:
This is done

gcoan, 2026-08-24:
> This is done

No it isn’t. Turning on debug mode, the file is still a YAML file format so can’t be directly attached to a github issue

[image]

gcoan, 2026-09-06:
fixed in #4941

closed 2026-09-06 by gcoan