springfall2008/batpred #677: Discharging 4-7pm not wanted

naltsta, 2024-01-30:
**Describe the bug**
For the last few days I have had discharges in my plan (great - I've had some cheap import) but with a constant 15p/kWh export they could happen any time. I now have a baseline export for any likely future saving sessions.
**Expected behavior**
If export is profitable do it just before importing for cheap (or a way to avoid exporting 4-7)
**Predbat version**

Auto updating

**Environment details**

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

naltsta, 2024-01-30:
Thanks Trefor - is it rate_export_override or rates_export_override ? Both seem to appear in the apps.yaml

naltsta, 2024-01-30:
And just to check...

Any future saving sessions will override my override right?!

springfall2008, 2024-01-30:
Yes it’s certainly with the S

actually good point, saving sessions won’t overrride the export overrride so this probably isn’t. A good fix

naltsta, 2024-01-30:
[image]

FYI Examples in comments don't seem to have the s

But guess I won't go down that route anyway!

springfall2008, 2024-01-31:
Okay I think I have released a solution now, please update to 7.15.8:

https://github.com/springfall2008/batpred/releases/tag/v7.15.8

Then in apps.yaml set:

  # To improve on saving sessions avoid export during peak periods
  rates_export_override:
   -  start: '17:00:00'
      end: '19:00:00'
      rate_increment: -10

This will reduce the export rate by 10p, which won't matter during saving sessions.

Sycamore967, 2024-01-31:
I've updated and set the override in apps.yaml.  I'm seeing this in the plan
[image]

should it say NoDischarge and not NoCharge?

Also the costs seem wrong they are going up - so it looks like PreBat is stopping discharge completely, not just a forced discharge.

springfall2008, 2024-01-31:
Looks like you set it on import not export?

Sycamore967, 2024-01-31:
Dam it ! - you are spot on.

Thank you.  I didn't realise that I'd need to create the actual override and had searched for an existing section in the yaml.
Still learning :)

springfall2008, 2024-01-31:
Hopefully that solves it now :)

naltsta, 2024-02-01:
Thank you - working for now so will double check it for best saving session

naltsta, 2024-02-01:
@springfall2008  worked beautifully today and showing 5p right now but tomorrow is showing as 15p still during the override slot. Will that change as it gets closer?

springfall2008, 2024-02-01:
> @springfall2008 worked beautifully today and showing 5p right now but tomorrow is showing as 15p still during the override slot. Will that change as it gets closer?

Probably, what’s in your apps.yaml?

RobinCu, 2024-02-01:
> @springfall2008 worked beautifully today and showing 5p right now but tomorrow is showing as 15p still during the override slot. Will that change as it gets closer?

I’ve had it in the apps.yml (as per Trefor’s code above). I can confirm that it shows as 15p for tomorrow but updates to 5p once midnight passes.

springfall2008, 2024-02-01:
I've made a fix to make tomorrow show correct: https://github.com/springfall2008/batpred/releases/tag/v7.15.10

closed 2024-02-03 by springfall2008