springfall2008/batpred #684: Please add support for Huawei inverters.

JohanAlvedal, 2024-02-01:
Please add support for Huawei inverters.

I use https://github.com/wlcrs/huawei_solar. 
This integration exposes the information and functions made available by Huawei Solar inverters directly via one of its Modbus interfaces in Home Assistant.

springfall2008, 2024-02-05:
Can you attempt to take a sample configuration from 'templates' and try it with your sensors? I suspect you will need my help to control charging and discharging, its not obvious to me how these are triggered

JohanAlvedal, 2024-02-05:
Taken from the github link above:
The integration offers multiple [services](https://www.home-assistant.io/docs/scripts/service-calls/) that you can call to dynamically change the behavior of the battery:

forcible_charge: Forcible Charge
forcible_charge_soc: Forcible Charge to a SoC level
forcible_discharge: Forcible Discharge
forcible_discharge_soc: Forcible Discharge to a SoC level
stop_forcible_charge: Stop the forcible charge or discharge
Forcible Charge

This allows you to forcibly charge your battery for a given duration or to a given SoC level of your battery. While a forcible charge is active, it will direct all power received from the solar panels to your battery instead of covering your home electricity usage.

If the "Charge from grid" toggle is set, it will also draw power from the grid until the given charge power level is reached.

Forcible Discharge

This allows you to forcibly discharge your battery for a given duration or to a given SoC level of your battery. This forces the inverter to inject more power to the AC-side than your home electricity usage, which will consequently be pushed onto the grid.

Stop Forcible charge or discharge

Calling this service will cancel any running forcible charge of forcible discharge command.

I have a reasonably functioning config. there will be no direct errors in the log as far as I can see. then of course it could be the wrong sensors I have, etc.

https://pastebin.pl/view/a70a067e

springfall2008, 2024-02-05:
I've made some changes on a branch:

1. Updated template - https://github.com/springfall2008/batpred/blob/huawei/templates/huawei.yaml

This has a new inverter type select HU and the services for charge/discharge defined

2. Update predbat.py - https://github.com/springfall2008/batpred/blob/huawei/apps/predbat/predbat.py

Added code for the new inverter type and using the services.

Please give this a try, however for it to work you need to fix your load sensor otherwise its not going to do anything useful

JohanAlvedal, 2024-02-05:
Thank you very much, I will double check and get back to you.

JohanAlvedal, 2024-02-05:
`2024-02-05 21:50:03.612180 INFO pred_bat: Inverter 0 Stop charge via Service
2024-02-05 21:50:03.616484 INFO pred_bat: Inverter 0 Stop discharge via Service
2024-02-05 21:50:03.627270 WARNING HASS: Error calling Home Assistant service default/huawei_solar/stop_forcible_charge
2024-02-05 21:50:03.627592 WARNING HASS: Code: 500, error: 500 Internal Server Error

Server got itself in trouble
2024-02-05 21:50:03.637760 WARNING HASS: Error calling Home Assistant service default/huawei_solar/stop_forcible_charge
2024-02-05 21:50:03.638070 WARNING HASS: Code: 500, error: 500 Internal Server Error

Server got itself in trouble
2024-02-05 21:50:03.647732 INFO pred_bat: Will not recompute the plan, it is 5.0 minutes old and max age is 10 minutes
2024-02-05 21:50:03.648596 INFO pred_bat: IBoost model today updated to 0.364
2024-02-05 21:50:03.649539 INFO pred_bat: Completed run status Idle`

springfall2008, 2024-02-05:
I must have the service name wrong, you can change it in apps.yaml if you can figure out the correct name

springfall2008, 2024-02-06:
Okay I think I might know the issue, the service needs a device_id to be set.

Can you:

1. Update predbat.py from the latest on the branch (https://github.com/springfall2008/batpred/blob/main/apps/predbat/predbat.py)
2. add 'device_id: xxxxx' to your apps.yaml where the xxxxx is the device_id that Home Assistant reports. The best way to find it is to go into developer tools, services and try to trigger the stop charge service, then once it works to go to YAML mode and copy the device_id from there. Also please snapshot what you see there in case my attempt still doesn't work.

JohanAlvedal, 2024-02-06:
Did you remove "HU" from the latest predbat.py ?

springfall2008, 2024-02-06:
Sorry I gave you the wrong link you need to take predbat from the Huawei branch and not
From main

gcoan, 2024-02-06:
> however for it to work you need to fix your load sensor otherwise its not going to do anything useful

@JohanAlvedal if you don't have a native 'home load' sensor easily available in HA, there's the code for custom template to calculate home load from import, export, solar and battery data in the HA docs you could utilise
https://springfall2008.github.io/batpred/apps-yaml#data-from-givtcp

JohanAlvedal, 2024-02-06:
> @JohanAlvedal if you don't have a native 'home load' sensor easily available in HA, there's the code for custom template to calculate home load from import, export, solar and battery data in the HA [...]

Made one now so I have to wait for some history.

springfall2008, 2024-02-06:
@JohanAlvedal did you try the correct code here? with device_id set https://github.com/springfall2008/batpred/blob/huawei/apps/predbat/predbat.py

JohanAlvedal, 2024-02-06:
> @JohanAlvedal did you try the correct code here? with device_id set https://github.com/springfall2008/batpred/blob/huawei/apps/predbat/predbat.py

Result of newest prebat.py

https://pastebin.pl/view/9a7685c4

springfall2008, 2024-02-06:
You would need to move away from monitor mode and into control charge to see if it does anything

JohanAlvedal, 2024-02-06:
> You would need to move away from monitor mode and into control charge to see if it does anything

https://pastebin.pl/view/025b0676

springfall2008, 2024-02-06:
Hmm, odd error - can you go to 'settings, system, logfiles' and select appdeamon (or adddeamon-predbat) and then check for the full error at the end of this log and paste it here please?

JohanAlvedal, 2024-02-06:
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/_base.py", line 456, in result
    return self.__get_result()
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/_base.py", line 401, in __get_result
    raise self._exception
  File "/usr/lib/python3.11/site-packages/appdaemon/adapi.py", line 1752, in call_service
    self._check_service(service)
  File "/usr/lib/python3.11/site-packages/appdaemon/adapi.py", line 1579, in _check_service
    if service.find("/") == -1:
       ^^^^^^^^^^^^
AttributeError: 'dict' object has no attribute 'find'

2024-02-06 21:35:03.822087 WARNING pred_bat: ------------------------------------------------------------
2024-02-06 21:40:03.724586 WARNING pred_bat: ------------------------------------------------------------
2024-02-06 21:40:03.724732 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-02-06 21:40:03.724837 WARNING pred_bat: Worker Ags: {'id': '2f9cb91f26b34c9f8125f68e37b1abdf', 'name': 'pred_bat', 'objectid': '52bbd711c5934b25b04ddaff81ad2496', 'type': 'scheduler', 'function': <bound method PredBat.run_time_loop of <predbat.PredBat object at 0x7f3af340f910>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 300, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-02-06 21:40:03.724926 WARNING pred_bat: ------------------------------------------------------------
2024-02-06 21:40:03.726075 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 12003, in run_time_loop
    raise e
  File "/config/apps/predbat.py", line 11999, in run_time_loop
    self.update_pred(scheduled=True)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 11037, in update_pred
    status, status_extra = self.execute_plan()
                           ^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 9841, in execute_plan
    inverter.adjust_charge_immediate(self.charge_limit_percent_best[0])
  File "/config/apps/predbat.py", line 2452, in adjust_charge_immediate
    self.base.call_service(self.base.get_arg("charge_start_service"), device_id=self.base.get_arg("device_id", index=self.id), target_soc=target_soc)
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 231, in inner_sync_wrapper
    f = run_coroutine_threadsafe(self, coro(self, *args, **kwargs))
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 313, in run_coroutine_threadsafe
    result = future.result(self.AD.internal_function_timeout)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/_base.py", line 456, in result
    return self.__get_result()
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/_base.py", line 401, in __get_result
    raise self._exception
  File "/usr/lib/python3.11/site-packages/appdaemon/adapi.py", line 1752, in call_service
    self._check_service(service)
  File "/usr/lib/python3.11/site-packages/appdaemon/adapi.py", line 1579, in _check_service
    if service.find("/") == -1:
       ^^^^^^^^^^^^
AttributeError: 'dict' object has no attribute 'find'

2024-02-06 21:40:03.726209 WARNING pred_bat: ------------------------------------------------------------
2024-02-06 21:45:03.762781 WARNING pred_bat: ------------------------------------------------------------
2024-02-06 21:45:03.763146 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-02-06 21:45:03.763319 WARNING pred_bat: Worker Ags: {'id': '2f9cb91f26b34c9f8125f68e37b1abdf', 'name': 'pred_bat', 'objectid': '52bbd711c5934b25b04ddaff81ad2496', 'type': 'scheduler', 'function': <bound method PredBat.run_time_loop of <predbat.PredBat object at 0x7f3af340f910>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 300, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-02-06 21:45:03.763519 WARNING pred_bat: ------------------------------------------------------------
2024-02-06 21:45:03.764956 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 12003, in run_time_loop
    raise e
  File "/config/apps/predbat.py", line 11999, in run_time_loop
[... 5 lines cut ...]
    status, status_extra = self.execute_plan()
                           ^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 9841, in execute_plan
    inverter.adjust_charge_immediate(self.charge_limit_percent_best[0])
  File "/config/apps/predbat.py", line 2452, in adjust_charge_immediate
    self.base.call_service(self.base.get_arg("charge_start_service"), device_id=self.base.get_arg("device_id", index=self.id), target_soc=target_soc)
[... cut ...]
  File "/usr/lib/python3.11/concurrent/futures/_base.py", line 401, in __get_result
    raise self._exception
  File "/usr/lib/python3.11/site-packages/appdaemon/adapi.py", line 1752, in call_service
    self._check_service(service)
  File "/usr/lib/python3.11/site-packages/appdaemon/adapi.py", line 1579, in _check_service
    if service.find("/") == -1:
       ^^^^^^^^^^^^
AttributeError: 'dict' object has no attribute 'find'

2024-02-06 21:45:03.765711 WARNING pred_bat: ------------------------------------------------------------

springfall2008, 2024-02-06:
What do you have in apps.yaml for 'charge_start_service' ?

JohanAlvedal, 2024-02-06:
charge_start_service:
    service: huawei_solar/forcible_charge_soc
    data:
      target_soc: 100
      power: "3000"

JohanAlvedal, 2024-02-06:
[image]

springfall2008, 2024-02-06:
No sorry in apps.yaml it should say:

charge_start_service: service: huawei_solar/forcible_charge_soc
device_id: XXXX

And you need to replace the XXXX with your device ID as per the above template

springfall2008, 2024-02-06:
Sorry I'm maybe not being specific, the apps.yaml should say:

#Services to charge/discharge
device_id: 6893ee838a0972bd3e50d9e0faa0ba5a

charge_start_service: huawei_solar/forcible_charge_soc
charge_stop_service: huawei_solar/stop_forcible_charge
discharge_start_service: huawei_solar/forcible_discharge_soc
discharge_stop_service: huawei_solar/stop_forcible_charge

JohanAlvedal, 2024-02-07:
predbat.log 
https://pastebin.pl/view/06a3d469

Addon log.
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: appdaemon-predbat
 Predbat pre-install in AppDaemon
-----------------------------------------------------------
 Add-on version: 1.0.6
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.5  (amd64 / generic-x86-64)
 Home Assistant Core: 2024.1.6
 Home Assistant Supervisor: 2024.01.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-appdaemon: starting
s6-rc: info: service init-appdaemon successfully started
s6-rc: info: service appdaemon: starting
s6-rc: info: service appdaemon successfully started
s6-rc: info: service legacy-services: starting
[00:49:32] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started

HA core log 
Logger: homeassistant.helpers.service
Source: helpers/service.py:246
First occurred: 01:17:09 (8 occurrences)
Last logged: 06:00:41

Referenced entities switch.predbat_load_filter_modal are missing or not currently available
Referenced entities switch.predbat_set_status_notify are missing or not currently available
Referenced entities switch.predbat_set_inverter_notify are missing or not currently available
Referenced entities switch.predbat_set_reserve_enable are missing or not currently available

springfall2008, 2024-02-07:
I think I might have to add 'power', if so what value?

JohanAlvedal, 2024-02-07:
> I think I might have to add 'power', if so what value?

Charge night time 4500w
Charge day 2500w
 Discharge 1500-2000w

JohanAlvedal, 2024-02-07:
[image]
[image]
[image]
[image]
[image]

springfall2008, 2024-02-07:
Do the above all work okay when you hit 'call service'?

It seems that we need to add power, what's the maximum power?

springfall2008, 2024-02-07:
Updated to include power setting here: https://raw.githubusercontent.com/springfall2008/batpred/huawei/apps/predbat/predbat.py

JohanAlvedal, 2024-02-07:
Yes, it works. But they need different types of settings in battery so they can be difficult to combine. Ex TOU only works if you have chosen it. Maximise ... only work at force….

JohanAlvedal, 2024-02-07:
> Updated to include power setting here: https://raw.githubusercontent.com/springfall2008/batpred/huawei/apps/predbat/predbat.py

Where do i add soc and watts?

JohanAlvedal, 2024-02-08:
It seems to work roughly, but there is still a little that needs to be polished as, for example, my house's fuses are at risk of blowing if it charges more than 2500w during the day unless the sun is shining / charging. It is currently a bit useless as tomorrow's prices still show today's and ?
So it doesn't always schedule a charge at night, which it might not always but still.
A perhaps easier way to go might be TOU. ex. Huawei Solar: Set TOU Periods
Time/duration.
in combination with the battery's work mide, Maximize self consumption. for sunny days when it can then charge the battery and then sell the surplus.
We've got 30 cm of snow here now so my solar cells give 0w at the moment so I'll let it roll for a few days and we'll see how it behaves.

[image]
[image]

springfall2008, 2024-02-08:
Now we have a first draft of the controls we need to get your basic Predbat predictions working correctly. Do you now have enough history to get sensible load predictions? What do your charts and battery plan look like?

Maximum charge rate can be capped in apps.yaml if you want

springfall2008, 2024-02-08:
How big is your battery, the charge and discharge seem very slow?

JohanAlvedal, 2024-02-08:
> How big is your battery, the charge and discharge seem very slow?

10 kw

springfall2008, 2024-02-08:
There's something wrong here, your battery is being reported as 100kWh

2024-02-07 06:00:03.089884 INFO pred_bat: Inverter 0 SOC: 96.0 kW 96 % Current charge rate 4500.0 w Current discharge rate 4500.0 w Current power 0.0 w Current voltage 52.0

springfall2008, 2024-02-08:
soc_max should be set to the battery size in kWh and not to a percentage number

soc_max:
  - 10

Should work if your battery is 10kWh 

On the other hand charge_limit should be set to the correct target charge %, although its not being used in this integration maybe it could be this: number.battery_end_of_charge_soc

So it should be:

```yaml
  soc_max:
   - 10
  charge_limit:
    - number.battery_end_of_charge_soc
```

JohanAlvedal, 2024-02-08:
Ok that explains a lot

JohanAlvedal, 2024-02-08:
but should I use 
soc_percent: - sensor.battery_state_of_capacity 
Or
soc_kw: - sensor.battery_to_kwh

springfall2008, 2024-02-08:
I believe either is fine, I'd use whichever is the primary source of information

JohanAlvedal, 2024-02-08:
It was supposed to start charging now at 22:00 but nothing happened.
Was there something I needed to do with power and soc in my Apps.yaml?

springfall2008, 2024-02-08:
What does the log say?

JohanAlvedal, 2024-02-08:
> What does the log say?

https://justpaste.it/cuug6

springfall2008, 2024-02-09:
So the force charge service didn't work

2024-02-08 22:00:03.682763 WARNING HASS: Error calling Home Assistant service default/huawei_solar/forcible_charge_soc
2024-02-08 22:00:03.683995 WARNING HASS: Code: 400, error: 400: Bad Request

springfall2008, 2024-02-09:
Can you update to this version with some extra debug info and re-run and gather the log?

https://raw.githubusercontent.com/springfall2008/batpred/huawei/apps/predbat/predbat.py

BTW: Did the change to the soc_max fix the plan, what does it look like now?

JohanAlvedal, 2024-02-09:
> BTW: Did the change to the soc_max fix the plan, what does it look like now?

Yes, that seems to be true now.

JohanAlvedal, 2024-02-09:
I have to take a step back and go through which working mode it should be, etc. because right now I see it's behaving a bit funny. Will see this weekend if I have time to test.

JohanAlvedal, 2024-02-09:
> The previous log showed the charge service failing, I added some more debug in the above link need to try to capture the error again so we can see why.

https://justpaste.it/g6buw

JohanAlvedal, 2024-02-10:
Hi springfall2008
I think I have found a possible way to go, which is much easier than force charge etc.
If you choose Work mode on the inverter, you can "force" it to do things quite easily, i.e. TOU, and if you set it to charge between 00:00-23:59, it always charges when you choose it.

However, if you choose Maximize self consumption, it starts to discharge, but only if it is a value in how many W (number.battery_maximum_discharging_power).

This means that when it is

Idle 1 = [select.battery_working_mode]: maximise_self_consumption, number.battery_maximum_discharging_power, 0W.

Idle 2 =  [select.battery_working_mode]: time_of_use_luna2000, number.battery_maximum_discharging_power, 0W.

Charge = [select.battery_working_mode] : time_of_use_luna2000, number.battery_maximum_discharging_power, 2500W

Discharge = [select.battery_working_mode]: maximise_self_consumption, number.battery_maximum_discharging_power, 1500W

Discharge
action:
  - service: select.select_option
    target:
      entity_id: select.battery_working_mode
    data:
      option: maximise_self_consumption
  - service: number.set_value
    target:
      entity_id: number.battery_maximum_discharging_power
    data:
      value: "1500"

Charge
action:
  - service: select.select_option
    target:
      entity_id: select.battery_working_mode
    data:
      option: time_of_use_luna2000
  - service: number.set_value
    target:
      entity_id:
        - number.battery_grid_charge_maximum_power
    data:
      value: "2500"

Idle 
action:
  - service: select.select_option
    target:
      entity_id: select.battery_working_mode
    data:
      option: maximise_self_consumption
  - service: number.set_value
    target:
      entity_id: 
         -number.battery_maximum_discharging_power
    data:
      value: "0"

or Idle 2 
action:
  - service: select.select_option
    target:
      entity_id: select.battery_working_mode
    data:
      option: time_of_use_luna2000
  - service: number.set_value
    target:
      entity_id: 
        -number.battery_maximum_discharging_power
    data:
      value: "0"

So if predbat wants to change these dynamically, it works fine for example with Battery charge/discharge curves.

action:
  - service: number.set_value
    target:
      entity_id:
        - number.battery_grid_charge_maximum_power
    data:
      value: "XXXX"

action:
  - service: number.set_value
    target:
      entity_id: 
         -number.battery_maximum_discharging_power
    data:
      value: "XXXX"

In other words, it can be difficult to make it work well via force charge if you don't always mix in the number.battery_maximum_charging_power / number.battery_maximum_discharging_power values because they control the power values.

For example huawei_solar.stop_forcible_charge is tricky to get that if you don't choose number.battery_maximum_charging_power / number.battery_maximum_discharging_power = 0 then it basically works immediately.

springfall2008, 2024-02-10:
I'm really struggling to understand what you mean here and why the maximum discharging power would be changed for charging.

I think the issue you might have right now is the huawei minimum charge/discharge level is 12% so you need to adjust **input_number.predbat_set_reserve_min** in Home Assistant to a value of 12.

JohanAlvedal, 2024-02-10:
On the contrary, to stop discharging, you must set number.battery_maximum_discharging_power to 0 if you have inverter mode: Maximize self consumption. Then to discharge the battery again, the value must be changed to, for example, 1500W

If you want, I can show you via google meet, so it might be easier to understand. However, my English is not 100%

springfall2008, 2024-02-17:
Surely setting discharging power to 0 will stop the battery from covering the house load or did I miss anything?

Did you try changing your reserve min to 12% as above and re-test?

JohanAlvedal, 2024-02-17:
Did you try changing your reserve min to 12% as above and re-test? = yes, I run now and have to wait for the result. but have been away for a couple of days when I ran a different flow then.

No, you didn't miss anything. What is expected to happen when it goes to idle? Should it continue to drain the battery? Because then my question is if there is any difference between discharege and idle?

I am a little unsure if my inverter's Maximize self consumption mode is the same as what you call eco mode. It discharges all the time as long as there is some % in the battery,

springfall2008, 2024-02-17:
Ah okay there is a terminology difference here, so 'idle' in Predbat means ECO Mode in GE inverter which is when the battery covers the house load. This is different from what it calls Discharge which is when you export to the grid at maximum rate.

JohanAlvedal, 2024-02-17:
the problem right now is that if it should charge to 33% as an example. Then it charges to 33%, stops charging, then automatically starts discharging for a while and then charges to 33% again, etc., etc. the only way to make the battery stop discharging is to set number.battery_maximum_discharging_power to 0.
Unfortunately, huawei_solar.stop_forcible_charge in combination with work mode Maximize self consumption does not help

springfall2008, 2024-02-18:
> the problem right now is that if it should charge to 33% as an example. Then it charges to 33%, stops charging, then automatically starts discharging for a while and then charges to 33% again, etc., [...]

That's progress, if its working then as the next step I can add this to the code :)

Did you have to change the apps.yaml any further?

JohanAlvedal, 2024-02-18:
Same thing with hold charge, there are two options as far as I can see. Either set (number.battery_maximum_discharging_power) to 0w or change workmode to (select.battery_working_mode) to time_of_use_luna2000 and then go to maximize_self_consumption to discharge the battery.

my apps.yaml 

https://justpaste.it/7iexm

JohanAlvedal, 2024-02-18:
> Same thing with hold charge, there are two options as far as I can see. Either set (number.battery_maximum_discharging_power) to 0w or change workmode to (select.battery_working_mode) to [...]

Or wait it's wrong with hold mode if you set it to always load in TOU (time_of_use_luna2000) then it's best to stay in Work mode: maximize_self_consumption and set number.battery_maximum_discharging_power to 0 W Then it should stay there

springfall2008, 2024-02-18:
I think your apps.yaml has an error, shouldn't charge/discharge rate be set as follows:

  charge_rate:
    - number.battery_maximum_charging_power
  discharge_rate:
    - number.battery_maximum_discharging_power

Have you tried turning off 'set_charge_freeze'? The reason it doesn't work is you have the reserve hard wired at 12 rather than set in a way that Predbat can change it. Normally in charge freeze the reserve is set to the current battery level to avoid the discharging, mosty as setting discharge power to 0 with GE inverters doesn't totally prevent discharge.

Can you try with set_charge_freeze off and also correct the above? Then please share a logfile?

JohanAlvedal, 2024-02-18:
> I think your apps.yaml has an error, shouldn't charge/discharge rate be set as follows:
>
> charge_rate: - number.battery_maximum_charging_power discharge_rate: - [...]

Aha let me change and restart.

JohanAlvedal, 2024-02-18:
https://justpaste.it/g2flr

[20 more replies]

closed 2025-08-21 by JohanAlvedal