springfall2008/batpred #686: Add ability to force Open Circuit Voltage (OCV) checks to self align the voltage to the correct SoC

mpartington, 2024-02-01:
If not regularly fully cycled, Giv batteries can build up a SoC error, ultimately resulting in a SoC drop to zero from a much higher percentage.  Whilst better post 3007 BMS, this can and does still happen.

Once batteries (on new BMS firmware) are empty and voltages have stabilised (typically 30 mins), they will perform Open Circuit Voltage (OCV) checks to self align the voltage to the correct SoC.  This is the current 'fix', however many of us don't regularly empty our batteries, so it doesn't run until there is a crash.

**Describe the solution you'd like**
As per the solution 'Willy Reckett' has posted on the GivTCP facebook page, it would be amazing if Predbat could also force this alignment automatically.

I see the solution being a 'calibration run' that could be scheduled for a set day or a simple toggle switch.  
 (1)  Predbat would fully discharge the batteries, aiming for the start of the off peak rate, with a small factor of safety say 15-30 mins into the slot.  
 (2) Predbat would then ensure scheduled charge is delayed (i.e. switched off), to allow the batteries to self level for approximately an hour.  
 (3)  Predbat would then switch the scheduled charge back on for the rest of the charge period

This would only really be suitable for people on Intelligent Go, or Go if you weren't fussed about getting a full charge (e.g. good solar forecast)

**Describe alternatives you've considered**
Using Willys Automation and switching Predbat to read only or Monitoring mode. 

**Additional context**
Example of an OCV correction, following a SoC drop of 20%
[image]

RobinCu, 2024-02-02:
+1! I'm currently doing this manually using a series of manual force discharges, but to have it automated would be fantastic.

mpartington, 2024-02-20:
I tried to adapt the automation from Willy for Go. Quite a lot of guess work how long each stage would take and some compromise to try and maximise the charge period (as in my view I'd rather run for at peak with low baseload for short period, to give battery maximum time to charge up - bigger reward vs risk).  Ironically I had another SoC drop earlier in the evening, so the 'health run' was definitely needed.

Another example of sequence in case this is implementable at some point

[image]

springfall2008, 2026-10-29:
Closing in favour of #2008, which asks for the same thing in more general terms: a scheduled deep discharge so the BMS can recalibrate SoC. Please add your +1 and any sequence details there so the discussion stays in one place.

In the meantime, the manual override selects (force export, then freeze or demand) can be driven from an HA automation to run a calibration cycle.

closed 2026-10-29 by springfall2008