springfall2008/batpred #687: Predbat plan displays incorrect import tariff during Savings Sessions

kamenmackay, 2024-02-02:
**Describe the bug**
Predbat plan displays incorrect import tariff during Savings Sessions

**Expected behavior**
I would expect that it would show the current import tariff for Octopus and the import tariff that the user is on

**Predbat version**

7.15.10

**Environment details**

- 3.6 kw Givenergy inverter and 9.5 kwh battery
- Standard HAOS installer on raspberry pi4

**Screenshots**
[image]

**Log file**
Attached
[predbat.log](https://github.com/springfall2008/batpred/files/14140089/predbat.log)

RobinCu, 2024-02-02:
The import tariff column reflects the cost of importing during a Saving Session.  Payments are made on net reduction, so if you import, the net impact will be 214.42p/kWh.

RobinCu, 2024-02-02:
For reference: https://github.com/springfall2008/batpred/issues/387

kamenmackay, 2024-02-02:
I read through the comments on that ticket and I think this is over-complicating matters by calculating the net impact and then using that as a figure. To me, if I did import during this period, Octopus would not charge me 214p/kwh but 38p/kwh or whatever the current import rate is for my tariff.

RobinCu, 2024-02-02:
The payment depends on the baseline (10-day average), so any import has a cost based on a comparison to the baseline.

Here are a few scenarios:
[image]

As you can see in scenario 2, an import of 1kWh reduces the payout by £2.30

That's been my experience over the last 9 sessions, including when IOG decided to charge my car for 20 minutes during a saving session!

springfall2008, 2024-02-02:
Effectively Predbat doesn't have any knowledge of saving sessions, so we just fake the import and export rates to make it avoid imports and plan exports. The rate amount is important so that Predbat can cost if its worth doing or not.

An enhancement maybe to show the real rates on the plan and highlight that its a saving session but most users seem happy with the current implementation.

gcoan, 2024-02-02:
It can be a bit confusing but I think the current solution works fine

And it is explained in the docs https://springfall2008.github.io/batpred/energy-rates/#octopus-saving-sessions

springfall2008, 2024-02-04:
I've made this clearer in the latest release v7.15.13 by showing a dollar sign on the rates that are adjusted for Octopus saving sessions.

closed 2024-02-04 by springfall2008