springfall2008/batpred #688: Nordpool

JohanAlvedal, 2024-02-02:
Hello
This seems like an interesting project. However, I have nordpool (tibber Sweden) as an integration in HA which delivers my electricity prices. Also has a Huawei inverter that communicates using modbus from HA. The question I have is if you can somehow get it in instead of octopus?

It would be nice to have these to work  with Nordpool
metric_octopus_import: 're:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_current_rate)'
metric_octopus_export: 're:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_export_current_rate)'

sensor.nordpool_kwh_se3_sek_3_10_025 shows the current price and have all data current tomorrow and so on as attribute. 
```

`Average  0,32
Off peak 1  0,25
Off peak 2  0,07
Peak  0,46
Min -0
Max  0,85
Mean  0,28
Unit  kWh
Currency  SEK
Country  Sweden
Region  SE3
Low price false
Price percent to average  1,01
Today  0,11, 0,13, 0,13, 0,15, 0,17, 0,18, 0,29, 0,82, 0,85, 0,82, 0,8, 0,61, 0,35, 0,33, 0,32, 0,33, 0,3, 0,29, 0,27, 0,21, 0,16, 0,11, 0,01, −0
Tomorrow  −0,01, −0,02, −0,03, −0,03, −0,03, −0,03, −0,01, −0, 0, 0,03, 0,03, 0, 0, −0, −0, 0,01, 0,05, 0,15, 0,15, 0,05, 0, 0,02, 0, −0
Tomorrow valid  true
Raw today
- start: '2024-02-02T00:00:00+01:00'
end: '2024-02-02T01:00:00+01:00'
value: 0.111
- start: '2024-02-02T01:00:00+01:00'
end: '2024-02-02T02:00:00+01:00'
value: 0.127
- start: '2024-02-02T02:00:00+01:00'
end: '2024-02-02T03:00:00+01:00'
value: 0.129
- start: '2024-02-02T03:00:00+01:00'
end: '2024-02-02T04:00:00+01:00'
value: 0.147
- start: '2024-02-02T04:00:00+01:00'
end: '2024-02-02T05:00:00+01:00'
value: 0.166
- start: '2024-02-02T05:00:00+01:00'
end: '2024-02-02T06:00:00+01:00'
value: 0.184
- start: '2024-02-02T06:00:00+01:00'
end: '2024-02-02T07:00:00+01:00'
value: 0.287
- start: '2024-02-02T07:00:00+01:00'
end: '2024-02-02T08:00:00+01:00'
value: 0.818
- start: '2024-02-02T08:00:00+01:00'
end: '2024-02-02T09:00:00+01:00'
value: 0.847
- start: '2024-02-02T09:00:00+01:00'
end: '2024-02-02T10:00:00+01:00'
value: 0.822
- start: '2024-02-02T10:00:00+01:00'
end: '2024-02-02T11:00:00+01:00'
value: 0.803
- start: '2024-02-02T11:00:00+01:00'
end: '2024-02-02T12:00:00+01:00'
value: 0.607
- start: '2024-02-02T12:00:00+01:00'
end: '2024-02-02T13:00:00+01:00'
value: 0.347
- start: '2024-02-02T13:00:00+01:00'
end: '2024-02-02T14:00:00+01:00'
value: 0.33
- start: '2024-02-02T14:00:00+01:00'
end: '2024-02-02T15:00:00+01:00'
value: 0.324
- start: '2024-02-02T15:00:00+01:00'
end: '2024-02-02T16:00:00+01:00'
value: 0.325
- start: '2024-02-02T16:00:00+01:00'
end: '2024-02-02T17:00:00+01:00'
value: 0.297
- start: '2024-02-02T17:00:00+01:00'
end: '2024-02-02T18:00:00+01:00'
value: 0.294
- start: '2024-02-02T18:00:00+01:00'
end: '2024-02-02T19:00:00+01:00'
value: 0.266
- start: '2024-02-02T19:00:00+01:00'
end: '2024-02-02T20:00:00+01:00'
value: 0.206
- start: '2024-02-02T20:00:00+01:00'
end: '2024-02-02T21:00:00+01:00'
value: 0.157
- start: '2024-02-02T21:00:00+01:00'
end: '2024-02-02T22:00:00+01:00'
value: 0.105
- start: '2024-02-02T22:00:00+01:00'
end: '2024-02-02T23:00:00+01:00'
value: 0.011
- start: '2024-02-02T23:00:00+01:00'
end: '2024-02-03T00:00:00+01:00'
value: -0.0
Raw tomorrow
- start: '2024-02-03T00:00:00+01:00'
end: '2024-02-03T01:00:00+01:00'
value: -0.014
- start: '2024-02-03T01:00:00+01:00'
end: '2024-02-03T02:00:00+01:00'
value: -0.021
- start: '2024-02-03T02:00:00+01:00'
end: '2024-02-03T03:00:00+01:00'
value: -0.025
- start: '2024-02-03T03:00:00+01:00'
end: '2024-02-03T04:00:00+01:00'
value: -0.029
- start: '2024-02-03T04:00:00+01:00'
end: '2024-02-03T05:00:00+01:00'
value: -0.029
- start: '2024-02-03T05:00:00+01:00'
end: '2024-02-03T06:00:00+01:00'
value: -0.026
- start: '2024-02-03T06:00:00+01:00'
end: '2024-02-03T07:00:00+01:00'
value: -0.006
- start: '2024-02-03T07:00:00+01:00'
end: '2024-02-03T08:00:00+01:00'
value: -0.001
- start: '2024-02-03T08:00:00+01:00'
end: '2024-02-03T09:00:00+01:00'
value: 0
- start: '2024-02-03T09:00:00+01:00'
end: '2024-02-03T10:00:00+01:00'
value: 0.03
- start: '2024-02-03T10:00:00+01:00'
end: '2024-02-03T11:00:00+01:00'
value: 0.03
- start: '2024-02-03T11:00:00+01:00'
end: '2024-02-03T12:00:00+01:00'
value: 0
- start: '2024-02-03T12:00:00+01:00'
end: '2024-02-03T13:00:00+01:00'
value: 0
- start: '2024-02-03T13:00:00+01:00'
end: '2024-02-03T14:00:00+01:00'
value: -0.0
- start: '2024-02-03T14:00:00+01:00'
end: '2024-02-03T15:00:00+01:00'
value: -0.0
- start: '2024-02-03T15:00:00+01:00'
end: '2024-02-03T16:00:00+01:00'
value: 0.007
- start: '2024-02-03T16:00:00+01:00'
end: '2024-02-03T17:00:00+01:00'
value: 0.048
- start: '2024-02-03T17:00:00+01:00'
end: '2024-02-03T18:00:00+01:00'
value: 0.145
- start: '2024-02-03T18:00:00+01:00'
end: '2024-02-03T19:00:00+01:00'
value: 0.146
- start: '2024-02-03T19:00:00+01:00'
end: '2024-02-03T20:00:00+01:00'
value: 0.049
- start: '2024-02-03T20:00:00+01:00'
end: '2024-02-03T21:00:00+01:00'
value: 0.001
- start: '2024-02-03T21:00:00+01:00'
end: '2024-02-03T22:00:00+01:00'
value: 0.02
- start: '2024-02-03T22:00:00+01:00'
end: '2024-02-03T23:00:00+01:00'
value: 0
- start: '2024-02-03T23:00:00+01:00'
end: '2024-02-04T00:00:00+01:00'
value: -0.0
Current price 0,32
Additional costs current hour 0
Price in cents false
```
`

JohanAlvedal, 2024-02-02:
[image]
[image]
[image]

springfall2008, 2024-02-02:
And you want to use the values from 'today' as your import rates?

JohanAlvedal, 2024-02-02:
Yes exactly and tomorrow if possible.
Export sensor is an equivalent but without VAT.

One question, if I don't want to use the function to charge an electric car, can I then uncomment all that apply to the car?

gcoan, 2024-02-02:
> 
> 
> Yes exactly and tomorrow if possible.
> 
> Export sensor is an equivalent but without VAT.
> 
> 
> 
> One question, if I don't want to use the function to charge an electric car, can I then uncomment all that apply to the car?
> 
> 
> 
> 
> 
> [image]

Just set num_cars to 0 in apps.yaml and predbat won't do anything with car charging

springfall2008, 2024-02-03:
I've submitted a change to MAIN which should now read this format of data, just point your metric_octopus_import/metric_octopus_export to the sensors and let me know if it works.

JohanAlvedal, 2024-02-03:
I'm not sure how?
like this?
  metric_octopus_import: 're:(sensor.(sensor.nordpool_kwh_se3_sek_3_10_025)'
  metric_octopus_export: 're:(sensor.nordpool_moms)'
or 
   metric_octopus_import: 
     - sensor.nordpool_kwh_se3_sek_3_10_025
   metric_octopus_export: 
     - sensor.nordpool_moms

  And if there is anything else I need to do?

springfall2008, 2024-02-03:
Like this I think

metric_octopus_import:sensor.nordpool_kwh_se3_sek_3_10_025
metric_octopus_export: sensor.nordpool_moms

JohanAlvedal, 2024-02-03:
`2024-02-03 18:00:00.204989 INFO pred_bat: Found 11520 load_today datapoints going back 1 days
2024-02-03 18:00:01.702497 INFO pred_bat: Car charging hold True threshold 6.0
2024-02-03 18:00:01.703405 INFO pred_bat: Current data so far today: load 22.3 kWh import 22.35 kWh export 0.0 kWh pv 23.34 kWh
2024-02-03 18:00:01.704290 INFO pred_bat: Fetching futurerate data from None
2024-02-03 18:00:01.705970 INFO pred_bat: WARN: No Octopus data in sensor sensor.nordpool_kwh_se3_sek_3_10_025 attribute 'all_rates'
2024-02-03 18:00:01.706860 INFO pred_bat: Error: metric_octopus_import is not set correctly or no energy rates can be read
2024-02-03 18:00:01.713997 INFO pred_bat: ERROR: Exception raised `

springfall2008, 2024-02-03:
It doesn't look like you updated to the version on 'main'?

JohanAlvedal, 2024-02-03:
ok but how do i do that?

springfall2008, 2024-02-03:
Select the version from the select.predbat_update drop down menu.

This code actually ended up released now also

JohanAlvedal, 2024-02-03:
it works but the price is not correct, for example now at 21:00 the price is 0.02 sec (extremely cheap).

[image]

springfall2008, 2024-02-03:
What should the price be according to the data, is it out by a certain factor?

JohanAlvedal, 2024-02-03:
the price from Nordpool is correct but it is shown in kronor equivalent to pound not pence. 0.02 SEK (1 GBP = 13.36 SEK 133.6 öre (pence). But maybe it should be? it is pence in import and export but in GBP in the far right column.

springfall2008, 2024-02-03:
Yes it’s pence on the left and pounds on the right for the total

gcoan, 2024-02-03:
> the price from Nordpool is correct but it is shown in kronor equivalent to pound not pence. 0.02 SEK (1 GBP = 13.36 SEK 133.6 öre (pence). But maybe it should be? it is pence in import and export but in GBP in the far right column.

Predbat assumes all the prices from the energy sensors are in pence and produces the plan in pence.
Just the total column on the right of the plan is pence x 100 ie GBP

Ignoring that the units are wrong for you, is the order of magnitude of everything correct?

I don't think we want predbat to start doing currency conversion but it may be possible to have a "display currency" setting or some Jinja post processing of the html plan to change the currency text strings

JohanAlvedal, 2024-02-03:
Well, that's right, I'm satisfied, very satisfied, thanks for the help so far. Now it will be allowed to run for a few days, then I will work with the settings a bit

closed 2024-02-04 by JohanAlvedal