springfall2008/batpred #689: Predbat modes, documentation query

gcoan, 2024-02-02:
Question on the GivEnergy forum from pacemaker:

"I found another issue in the docs:

https://springfall2008.github.io/batpred/what-does-predbat-do/#predbat-status

- Freeze discharging - The battery is covering the load but charging is disabled, thus any solar generated will be exported.
- Freeze charging - The current battery level is held and the grid/solar covers any load. Solar power will also be used to charge the battery.

The above is the wrong way round. Freeze Discharging stops the battery from discharging at all, grid/solar cover the house load."

I wanted to check with @springfall2008 before answering which was correct or not.

Personally I find the different Freeze statii a bit confusing anyway as they are named so similarly, so didn't want to give a POV on how they should operate

springfall2008, 2024-02-02:
While the terminology maybe  confusing it is correct. 

The way to think if it is that Freeze Charging is like a charge but to the current battery level while Freeze Discharge is like a discharge to the current level. 

In reality Freeze Discharging is just ECO mode with charging disabled, which causes solar to export.

gcoan, 2024-02-02:
Great thanks for a quick reply, I'll update the docs to try to explain it further

closed 2024-02-02 by gcoan