springfall2008/batpred #706: Disable Predbat setting battery charge power when calbriation is happening - also surface a message

bigsmelly, 2024-02-03:
**Is your feature request related to a problem? Please describe.**

I've had an issue where seemingly a GivTCP corrupted command caused it to kick off a battery calibration.
Unforunately batpred was running and had decided to set discharge slots.

This meant that at some point during the charge part of the calibration, batpred set the charge power to 0w.
Which meant the calibration didn't seem to finish.

I solved this with stopping givtcp on my HA and resetting the charge powers on the inverter ( I think it may also set charge power itself semi-regularly to ensure the charge happens)

**Describe the solution you'd like**

Batpred to notice when a calibration is in progress and not attempt to control the system - i.e. set charge limits to default then do nothing until it finishes.

You can see this in the REST data as "soc_force_adjust"

   # battery calibration stages (0:off  1:start/discharge  2:set lower limit  3:charge
    # 4:set upper limit  5:balance  6:set full capacity  7:finish)

so for anything other than zero, it would be nice is batpred would pop up a message on the info bar .

**Additional context**

Hybrid gen1 inverter, 9.5 battery , firmware

Inverter Model
GIV-HY5.0
Software Version
D0.450-A0.451

Thanks!

springfall2008, 2024-02-04:
Fixed in v7.15.13

springfall2008, 2024-02-04:
Now Predbat will detect calibration, it will reset charge and discharge rates to full power if they are not already and wait.

bigsmelly, 2024-02-04:
Thank you for the quick turnaround : -)  thought it might be useful for others as well

closed 2024-02-04 by bigsmelly