springfall2008/batpred #707: Performance Issue with update_time_loop() Callback Execution Time

stefan-west1, 2024-02-03:
Hello,

I'm currently using BatPred in my Home Assistant setup, running on an Unraid server with AppDaemon in a Docker container. My hardware is quite capable, featuring 2x E5-2680 v4 processors, so I believe hardware limitations are unlikely to be the cause of my issue.

I've observed that certain interactions, such as toggling the "read only" switch from the Home Assistant UI, can take a few minutes to be reflected. Upon checking the AppDaemon logs, I found a warning indicating that the update_time_loop() callback in PredBat is taking significantly longer to complete than expected:

WARNING AppDaemon: Excessive time spent in callback 'update_time_loop() in pred_bat', Thread 'thread.thread-1' - now complete after 225.795194 seconds (limit=30.0)

Given AppDaemon's asynchronous nature and support for multi-threading, I'm curious:

Does PredBat explicitly support multi-threading or asynchronous execution to optimize performance for long-running tasks or IO-bound operations?

Could the performance issue I'm experiencing be mitigated by implementing multi-threading or optimizing the app's asynchronous execution patterns?

I'm eager to understand whether there's a way to enhance the app's responsiveness and reduce the execution time for such callbacks.

springfall2008, 2024-02-03:
Unfortunately it’s kind of as expected in the current implementation. I could certainly find a way to make the UI toggles change sooner but the state can’t be reflected in the output until a new run has been completed.

springfall2008, 2024-02-03:
The performance itself is solely bound by the time it takes to calculate the plan, which is really a linear search each with a simulation. I did wonder if I could run more than one simulation in parallel but python threads seem not to be real multi threading. Any ideas how I could make use of multiple CPUs?

gcoan, 2024-02-03:
Unfortunately Stefan it's part of the way predbat works

Predbat runs automatically every 5 minutes so what you see in home assistant is the sensors and plan for the most recent run. 

When you make a change to any of the home assistant switches, input numbers or selectors, predbat detects the change and starts a new run.

Depending on the length of your prediction plan and the optimisation options set it can take 1-3 minutes for a full predbat execution run.
That's why you see a delay in predbat responding to controls its because it's doing a full evaluation of the plan and cost optimisation and that can be cpu intensive.

You do sometimes see warnings in the log about excessive time, this is quite normal due to the time predbat can take. 

Multi threading or async processing is probably not going to materially change the run time.

Under normal circumstances the execution duration isn't an issue, just when you change a control it can seem slow to respond

springfall2008, 2024-02-04:
Actually I maybe wrong in terms of the events themselves if I change them to async then they update right away, but obviously the plan won't change until Predbat has time to update it.

springfall2008, 2024-02-04:
Can someone please try the version on 'main' which has this fix on it and see if it works okay for you?

gcoan, 2024-02-04:
I've tried the new version on main, it certainly seems a lot more responsive when changing multiple force charge/discharge slots for example, would select one, the change would then be almost instantaneous with the [] appearing around the new selected slot.
This does make it a lot easier to make multiple 'force' instructions to Predbat
Adding discharge and removing discharge worked perfectly as did adding charge, but removing charge, I selected [12:00:00] and its still not cancelling it
[image]

[image]

Not used idle in this test.

I wonder, and this is something I have thought before, whether we need a binary_sensor.predbat_calculating that indicates when predbat is calculating or not.  Would need to exclude it from history retention of course.  At the moment the only way to see whether the plan is up to date is to look at the end of the log file, and with async commands from the UI to predbat this will be more prevalent an isssue

springfall2008, 2024-02-04:
Interesting, I can't replicate your issue with it not clearing the setting. It maybe not 100% due to some race conditions, but it seems to work for me.

Release in v7.15.13, lets see what people find.

gcoan, 2024-02-04:
I've  done some more tests and its happened a couple of times now on both force charge and force idle.
Often when you select a time from the dropdown, the dropdown then clears and there's no entry shown selected in the selector.  When that happens, it all works fine, you can then select the same time in square brackets to cancel it, or you can select another time to force charge, etc.
But some of the time when you select a time from the dropdown, the control doesn't clear, and the time you have selected remains selected.  Then if you try to select the same time again to cancel it, even if you select the time in brackets, I think HA doesn't detect that the selector has been changed and so doesn't trigger an event in predbat.
So the issue is around the way that once a time is selected from the dropdown, what happens next, whether the selected time is 'cleared' so the dropdown shows a blank state, or whether it remains with the time you just selected or not

springfall2008, 2024-02-04:
There appears to be a bug in HA that sometimes means you have to reload the page to get the selector to show the correct value. I wonder if the two are related in some way. The blank is an example of this as I change the state to show the selected times but it shows blank unless you refresh the page. It maybe there’s a workaround to this, I’ll have a play when I get time

stefan-west1, 2024-02-04:
Sorry I've only just seen these replies. I've just updated to the latest version, and it's much better thank you. I found that sometimes Predbat didn't always want to charge when agile prices dropped below 0, so I made a quick automation to put Predbat into read only and force a manual charge. But before with the delay it was difficult to know if the automation had ran correctly or if something had timed out. But now it works perfectly. Thank you

springfall2008, 2024-02-04:
If you want you can just use the force dropdown to select the slots you want to charge in, but it may not always be obviously that much better.

springfall2008, 2024-02-04:
So I think I found the selector issue, basically HA doesn't update the selector correctly if you don't change its value. I've changed the displayed values to start with "+" to make them different from the drop down values and hence force it to update correctly.

7Van737, 2024-02-04:
I used the Force Charge options the other night and confirm the recent changes have made the function much faster and smoother!   Also like the time stamp on the HTML plan.

gcoan, 2024-02-04:
> So I think I found the selector issue, basically HA doesn't update the selector correctly if you don't change its value. I've changed the displayed values to start with "+" to make them different from the drop down values and hence force it to update correctly.

Thanks Trefor, I tried the new version and no issues with the selector not being updated now each and every time.  Perfect, and a lot more responsive too

closed 2024-02-04 by springfall2008