springfall2008/batpred #714: Repeated Errors during low power mode

dochtie, 2024-02-04:
**Describe the bug**
During low power mode my inverter often will not accept the charge value Predbat tries to set. This causes predbat to repeatedly try before finally reporting ‘charging with errors check log’.

2024-02-04 14:36:15.877554 INFO pred_bat: Inverter 0 current charge rate is 478 and new target is 625
2024-02-04 14:37:18.232641 INFO pred_bat: WARN: Inverter 0 set charge rate 625 via REST failed got 478

**Expected behavior**
I don’t know if there is a set of values my inverter will accept. If not then predbat should accept a rate within a certain range rather than keep trying to change it. 

**Predbat version**

7.15.12

**Environment details**

- All in One
- HAOS

**Screenshots**
[image]

**Log file**
[appdaemon 3.log](https://github.com/springfall2008/batpred/files/14157287/appdaemon.3.log)

springfall2008, 2024-02-04:
Fixed in v7.15.13

dochtie, 2024-02-04:
Within the hour 🤯

closed 2024-02-04 by dochtie