springfall2008/batpred #718: Nordpool, It shows the same price today and tomorrow but it doesn't work between 00:00 - 07:00

JohanAlvedal, 2024-02-04:
**Describe the bug**
Now when I start reviewing I see that after 00:00-07:00 it shows -0.10 ? so after 00:00 it is wrong. Once the times are shown, it is the same price as today's.
It shows the same price today and tomorrow but it doesn't work between 00:00 - 07:00
**Expected behavior**
It should show today's price and after 00:00 tomorrow's price.

**Predbat version**
7.15.12

[image]

[image]

springfall2008, 2024-02-04:
Can you show the matching data for this?

JohanAlvedal, 2024-02-04:
You can try installing nordpool in dim HA and see how it looks. No registration etc. is needed.
I have added everything to the sensor at the following link.
[https://pastebin.pl/view/a340ee94](url)

[image]

JohanAlvedal, 2024-02-05:
It's better than before but the same price as today after 00:00 (Tuesday)

Predbat v7.15.15

[image]
[image]

springfall2008, 2024-02-05:
Isn’t this because the nordpool data runs out and then predbat just repeats it?

JohanAlvedal, 2024-02-05:
Yes, it might be. But sometime after 13:00, tomorrow's prices are included.

springfall2008, 2024-02-05:
I'm struggling to understand if there is still an issue or if this is working as expected?

BTW: Predbat also has Nordpool access but it's used to adjust Octopus Agile data.

JohanAlvedal, 2024-02-05:
Hello
Yes, right now it's a problem if you expect to charge the battery after 00:00 predbat don't know what the real price is. I fully understand that this is not priority 1 for you, so right now I'm "playing" a bit with this. However, it would be fun if you could get it started as I know that there are probably some who would appreciate it if this got started with nordpool and Huawei inverters. I'm still struggling to get started with the charging schedule, nothing is happening yet. However, I think it may have to do with my sensor not having enough history yet. I'll read and check out your videos tonight. 
(Unfortunately, as far as I know, there is no Octopus Agile data for us Swedes)

springfall2008, 2024-02-09:
I figured it out now, the tomorrow data is under raw_tomorrow, will release a fix shortly.

springfall2008, 2024-02-09:
https://github.com/springfall2008/batpred/releases/tag/v7.15.19

closed 2024-02-10 by JohanAlvedal