springfall2008/batpred #720: Incorrect Cost Calculation in HTML Plan

JonathanLew1s, 2024-02-04:
**Describe the bug**
At 23:30 tonight the rate is £0.00, the plan somehow calculates the cost of charging in this slot as 47p.

**Predbat version**

7.15.13

**Screenshots**
[image]

**Log file**

[_appdaemon_logs-11.txt](https://github.com/springfall2008/batpred/files/14157903/_appdaemon_logs-11.txt)

springfall2008, 2024-02-04:
This is your standarding change!

JonathanLew1s, 2024-02-04:
Sorry.

closed 2024-02-04 by JonathanLew1s

gcoan, 2024-02-04:
JonathanI deleted the standing charge sensor in my apps.yaml so my Predbat plan matches my energy dashboard and only has consumption costs. Not much I can do to optimise the standing charge and I felt it distorted the data

JonathanLew1s, 2024-02-04:
Historically it's been commented out. Updated my apps.yaml last week, to align with upstream, and forgot about standing charge.

gcoan, 2024-02-04:
Ha ! 

Well if that's the only thing you got wrong with the apps.yaml, it's a good result