springfall2008/batpred #729: Manual Force Idle function does not pause discharge.

7Van737, 2024-02-05:
**Describe the bug**
The manual force idle function does not pause the battery from discharging - I expected that the battery would neither charge/discharge and that any load would be met by the grid.   

 The command is recorded in the AppDaemon log "2024-02-05 22:11:44.054567 INFO pred_bat: Manual override: manual idle times now ['02-05 22:00:00']", the Predbat status is reported as "Idle" and the Predbat plan shows a downward arrow (expected horizontal arrow??) and an upside down 'F' suggesting the command has been accepted but no record of the command is recorded in the GivTCP log.

I assume the 'idle' is to be achieved by switching 'Eco Mode' off - looking at the Enable Eco Mode register on the GivEnergy portal confirms that the Eco Mode remains switched on throughout the idle period.

**Predbat version**

v7.15.16 - but issue first noticed last night on earlier version 

**Environment details**
AIO, HAOS

RobinCu, 2024-02-05:
Idle will supply the load from the battery (or solar). The downward arrow represents the direction of the battery SOC.  See here for status descriptions:
https://springfall2008.github.io/batpred/what-does-predbat-do/#predbat-status

7Van737, 2024-02-05:
Thanks - understand.  

I think this functionality would be better called "Eco mode" to be consistent with the GivEnergy terminology - but understand this is so named to cater for other inverters.  

I also think it would be useful for a Manual Force Freeze Charge capability to be enabled (similar UI as for the Force Charge/Discharge/Idle options) - this is the functionality I thought "Idle" provided.

gcoan, 2024-02-05:
My expectation is that manual Idle would stop any current charging and discharging and set predbat to idle mode which corresponds to Eco mode in givenergy - battery charged from pv (excess exported) and battery supplies house load up to the inverter limits.

So I think it's working to plan.

Sounds like @7Van737 wants some sort of freeze mode?

gcoan, 2024-02-13:
Expanded the docs to explain what manual idle does more clearly