springfall2008/batpred #730: (Minor) missing Nordpool rate for 23:30pm

gcoan, 2024-02-06:
**Describe the bug**
Not sure if it is a formatting error or a genuinely missing rate, but with the change to see where future rates are being populated from, it appears that the 23:00 rate and 00:00 tomorrow night are both coming from Nordpool, but the 23:30 is coming from Predbat repeating today's rate

Not sure whether this is a formatting error in the plan or a loop issue in the norddata retrieval.  Can't see anything in the log that suggests the data is missing

Low priority issue

**Expected behavior**
23:30 rate populated from Norddata as are the 00:30, 01:30, etc, up to when they are populated by Octopus

**Predbat version**
7.15.16

**Screenshots**
[image]
[image]

gcoan, 2024-02-06:
Logfile if you need it
[appdaemon (25).log](https://github.com/springfall2008/batpred/files/14179699/appdaemon.25.log)

springfall2008, 2024-02-07:
It's certainly has a bug, need to figure out what causes it.

springfall2008, 2024-02-09:
Good spot, this was a nasty wrapping bug and also there was an issue with some of the rates getting overriden by the average. Hopefully better now

springfall2008, 2024-02-09:
https://github.com/springfall2008/batpred/releases/tag/v7.15.18

gcoan, 2024-02-09:
Thanks Trefor, I'll have a look tomorrow but am sure it'll be fixed.

Cheers

closed 2024-02-11 by springfall2008