springfall2008/batpred #744: Status Description confusing

pacemaker82, 2024-02-10:
Can we take a look at or confirm the following status description, as a layman it seems confusing or counter-intuitive. Example:

From: https://springfall2008.github.io/batpred/what-does-predbat-do/#predbat-status

> Freeze discharging
> This is the same as Idle (Eco) mode but with **charging disabled.** The battery is covering the load but as charging is disabled, any solar generated will be exported.

Bold for emphasis. If the status is "freeze discharging", my expectation is that the battery discharging is disabled, when in fact CHARGING is disabled, discharging **_is_** happening to the house and is not "frozen".

springfall2008, 2024-02-10:
I see where you are coming from, the idea is that freeze charging is when you charge but the charge level is frozen while freeze discharge is when you discharge but the discharge level is frozen. 

In freeze discharging the battery is in discharge mode, while in freeze charging the battery is in charge mode.

pacemaker82, 2024-02-10:
Thanks, so I have this in my plan:

[image]

You mentioned:
> freeze discharge is when you discharge but the discharge level is frozen.

As I dont see a limit set, how is the level frozen in this case?

springfall2008, 2024-02-10:
>As I dont see a limit set, how is the level frozen in this case?

Effectively its the current SOC as the limit, it means we only exoport excess solar rather than the battery.

gcoan, 2024-02-13:
I have updated the docs to expand the explanation of what the freeze modes do.

closed 2024-03-04 by springfall2008