springfall2008/batpred #748: Notification Spam

dandwhelan, 2024-02-11:
**Describe the bug**
I get notified very regularly about state changes even though I've the two switches toggled off in HA. I've tried toggling them on and off. 

It's been happening for a while and I feel other people would be complaining about this, so maybe it's just me. 

In an ideal world I only want to see when predbat has been updated and when I've joined saver sessions. 

[appdaemon (2).txt](https://github.com/springfall2008/batpred/files/14231805/appdaemon.2.txt)
[apps.txt](https://github.com/springfall2008/batpred/files/14231806/apps.txt)

**Expected behavior**
Have no notifications about state changes

**Predbat version**

7.15.20

**Environment details**

- Inverter and battery setup G2 in 5kw. 8.2kw battery
- Standard HAOS installer 
- Anything else?

I don't want to turn off all notifications as I've other ha notifications that are useful. 
**Screenshots**
[image]
[image]
[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

springfall2008, 2024-02-11:
Hopefully I found the issue and fixed it in the latest release, please re-test

dandwhelan, 2024-02-11:
Thanks, that's alot quieter.

Will I still get notifications about updates and saver sessions with those two options turned off?

springfall2008, 2024-02-11:
Yes correct

gcoan, 2024-02-11:
I'm.still seeing a lot of notifications.

I only have set status notify on and here's the most recent notifications, notice the multiple discharging notifications as the target changes and then swap between hold charging and idle at 4% - battery is exhausted so these are both doing nothing 

[image]

PianSom, 2024-02-11:
Feature request - it would be helpful to have different notification types directed to different notification services.

Example - ATM all my Predbat notifications are sent to a private Discord channel, where I can check them periodically. But it would be great to be able to divert eg Saving Session notifications to an iOS notifier

Nilogax, 2024-02-27:
I have status notify set On and inverter notify Off. I only want to be notified if the status actually changes, but current behaviour is that I still get notified if eg status is Idle, then there's a change to inverter settings (eg set discharge time window), then status reverts to Idle. I'd only expect to be notified if the status changed to something other than the last notified value.  Screenshot attached where I received 3 successive notifications that the status had changed to Idle. 

[image]

gcoan, 2024-03-11:
This morning, an unfortunately great example of notification spam.

The agile rates are flat/poor today so the predbat plan was to charge early in the morning and keep charge all day through to the peak. No force discharge or Idle at all.
[image]

Status notify on, inverter notify off

From when charging started last night
[image]

there is a hold charging notification every 5 minutes from predbat all through the night and still doing so at 9:30 this morning 
[image]

gcoan, 2024-03-11:
> This morning, an unfortunately great example of notification spam.
> 

> there is a hold charging notification every 5 minutes from predbat all through the night and still doing so at 9:30 this morning 

logfile for the above:
[appdaemon (26).log](https://github.com/springfall2008/batpred/files/14557358/appdaemon.26.log)

dandwhelan, 2024-04-14:
All good

closed 2024-04-14 by dandwhelan