springfall2008/batpred #751: Predbat 7.15.21 stuck in install state

pacemaker82, 2024-02-11:
**Describe the bug**
The installation of 7.15.21 is stuck on an installing state, have received two messages on my phone saying it installed, but predbat seems top be offline, the plan is null.

**Expected behavior**
clean installation

**Predbat version**

7.15.21 

**Environment details**

Home Assistant Green, 

**Screenshots**

[image]

[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

closed 2024-02-11 by pacemaker82

gcoan, 2024-02-11:
My 7.15.21 upgrade installed flawlessly and is running fine.

It looks like the upgrade process within HA has got stuck

Can you try shutting down the AppDaemon or Predbat-AppDaemon add-on (depending on what you are using, I suspect its the latter) and restart it, to see if that brings it back to life?

Maybe then try a full shutdown and restart of Home assistant and all the add-on's to clear the stuck status

pacemaker82, 2024-02-11:
The restart of the app daemon fixed it

gcoan, 2024-02-11:
Great, glad you were able to sort it out