springfall2008/batpred #752: Feature Request: Plan stability improvements

dochtie, 2024-02-12:
**Is your feature request related to a problem? Please describe.**
With borderline situations the plan can flick back and forth, causing unwanted behaviour in the current and next slots. Can cause setting and unsetting of reserves, full speed charging followed by hold charge and then full speed charging again.

**Describe the solution you'd like**
Add some fuzziness metric or some other way to make the current planned behaviours a bit more stable. EG: only alter the current or successive planned behaviour if it saves N (pence). Or some other way to make predbat a bit more committed its actions for the current 30 minute slot and maybe the next 30 minute slot… even if it costs a few more pence more.

**Describe alternatives you've considered**
turning off in day adjustment makes a big difference especially at the beginning of the day but then you lose this great feature. Adding more days and longer history into apps.yaml also helps. Unfortunately especially with agile there is always going to be borderline and unstable plans. 

**Additional context**

gcoan, 2024-02-12:
Have you tried input_number.predbat_metric_min_improvement (default value 0) and input_number.predbat_metric_min_improvement_discharge (default 0.1) which set minimum pence cost improvement thresholds?

I have mine set to 0 and 0.2 to try to dampen the plan down.  You could try much bigger values such as 0.3 or 0.5.

dochtie, 2024-02-12:
I have discharge set at 1p but left the charge at 0. I do want to encourage it to charge. I think it’s caused by tiny fluctuations in predicted load or the actual load being different than expected, nudging the plans back and forth. I would prefer predbat to have a slight weighting to its current action (whatever that may be). After all it can reevaluate in the next 30 min slot which isn’t far away.

dochtie, 2024-02-12:
Some more stickyness to a plan even if predicted to cost a little more.

chalfontchubby, 2026-09-16:
input_number.predbat_metric_min_improvement  is supposed to prevent fluctuations - there wasn't a comment to say whether this had been used.

Going to mark this for closure - feels like what you ask for is already there

gcoan, 2026-09-19:
> input_number.predbat_metric_min_improvement is supposed to prevent fluctuations - there wasn't a comment to say whether this had been used.

that wouldn't have been an option when this was raised in 2024, its a newer feature.  I think this gives what the requestor was asking for

closed 2026-09-19 by dochtie