springfall2008/batpred #754: Winter battery temperature optimisation

gcoan, 2024-02-13:
**Is your feature request related to a problem? Please describe.**
It's well known that battery performance degrades in cold weather with charging and discharging rates affected, ultimately to the point that batteries will refuse to charge or discharge if the temperature becomes too low.

Tesla Powerwalls circumvent this issue by means of in built battery heating.

GivEnergy are introducing a new portal control soon - see this video https://youtu.be/oChIkOkpf2I - that once enabled will perform a 30 minute force charge if the battery temperature gets too low.

I propose a similar mechanism being built into Predbat which means all battery control is under one banner rather than the portal instructing different things that Predbat knows nothing about.

**Describe the solution you'd like**
Make the solution generic and configurable to cover different battery types/limits.

In apps.yaml:
- battery low temperature config eg 2 degrees
- battery temperature sensor config (pointing to battery entity)
- battery low temperature charge duration eg 30 minutes
Potentially also charge rate or % of max rate config?

When Predbat recognises that the battery temperature is too low it initiates a charge of the battery for the specified duration to raise the temperature.  There will need to be some way of mapping battery sensors to inverters (for charging) to cover multiple battery and inverter scenarios, but TBH having this would also be useful for a path towards more specific charging for different batteries on different inverters (e.g. covering dissimilar battery sizes) as at the moment Predbat assumes all batteries are the same capacity and charge/discharge at the same rate

**Describe alternatives you've considered**
Own bespoke HA automation that puts predbat into read only if it detects low temp and manual charging required.

Use GivEnergy solution but this is likely to clash with Predbat if GE starts charging when Predbat has other ideas what the inverter should be doing.

7Van737, 2024-02-13:
At present GivTCP does not create an entity for the All-in-One battery temperature.   The temperature does exist in the 'raw' variables and a HA sensor can be created.  

If this winter temperature feature is to be added to Predbat it would therefore be useful to have the facility to add the created sensor to the apps.yaml (in the same way as the iboost_energy_today is added) as each AIO user is likely to have uniquely named created battery sensor.

gcoan, 2024-02-13:
I was assuming that the temperature sensor would need to be configurable and would enable extension to other non givenergy batteries as well