springfall2008/batpred #755: Best SoC Keep ignored later in the day

iainfogg, 2024-02-13:
**Describe the bug**
As I understand it, Best SoC Keep us supposed to set the level that Predbat won't plan to drop below, although it doesn't prevent your battery actually discharging below that level through house load.

I've always used this to set the level that Predbat should discharge to by 23:30 prior to the Intelligent cheap rate kicking in.

For ages this has worked well, but recently (on multiple versions) I've found that the plan early evening shows it discharging to l down to the level I set it to e.g 1.5 or 2.5kWh.

However, if I look at the plan around 22:30 or 23:00 it's now planning to discharge right down to 4-5% SoC, ignoring the limit I've set. You can see this in the screenshot below.

This results in the system running the battery right down, then the house load ends up coming from the grid.

On another ticket I did mention about all my settings getting wiped, so it's possible I've got something set wrong, but I'm not clear why the plan seems to evolve during the evening from respecting the Keep parameters to ignoring it.

Any ideas please?

If you need logs from a particular time just let me know.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Predbat version**

7.15.21, and also at least two earlier versions

**Screenshots**

[image]

**Log file**

I can grab this if needed, just can't do it right now.

gcoan, 2024-02-13:
You might be confusing best soc keep and best soc min?

Keep is the preferred minimum but it's a soft limit so can be exceeded.
Min is a hard limit that won't be exceeded.

Maybe you need to set best soc min to the limit you want to prevent grid import

springfall2008, 2024-02-14:
As it currently works 'keep' is ignored for the next 4 hours, meaning it only impacts longer term planning but has no impact on the short term.

best soc min would set the minimum charge level for all the slots, which only really works if you have a fixed charge window. 

I suppose the question is why does the house end up running on grid, is Predbat not calculating the load correctly?

It would certainly be possible to prevent discharge that violates best soc keep, that maybe the ask here?

iainfogg, 2024-02-14:
Thanks both.

I guess I'm not sure why 'keep' is used early in the day, then put aside for the next 4 hours. The start of day plan does what I expect, it plans to discharge down to the 'keep' level, but still with a level of buffer in case additional unplanned house load demands more.

As it is, it's using another target to discharge down to, and maybe due to calibration issues the battery is hitting its limit and stopping supplying the house 15-20 minutes before the charge period starts.

I think it's important that Predbat is not aiming to discharge all the way down to an enforced limit, because if a dishwasher or washing machine run happens, that's going to tip it over at the end of the day.

Personally I'd be keen on Predbat always respecting the 'keep' limit, but I'm interested in knowing what the reason for the 4h thing is?

springfall2008, 2024-02-14:
If the keep is respected now then whenever the battery falls below that level it would immediately charge, which kind of defeats the purpose.

iainfogg, 2024-02-14:
Also, I'm sure this used to work fine, when did the 4h change come in?

iainfogg, 2024-02-15:
Here's an example of what's happening in practice, this is the import between 23:00-23:30 yesterday.

[image]

iainfogg, 2024-02-16:
Do you have a suggestion @springfall2008 of how I can avoid it running down to a level where either the battery stops discharging, or where Predbat will start recharging it, before 23:30 please?

gcoan, 2024-02-16:
Can you share your predbat plan and the soc history for the same duration.

Usually the historic load should give a more than good indication of future load. What do you have days previous set to?

iainfogg, 2024-02-17:
Hi @gcoan I'm not sure specifically what you're after, as in the plan as at what point of the day? During the day when it looks ok, or later on when it's planning to discharge to 4-5%?

And any idea what's a good way to get the SoC history at corresponding 30 minute intervals?

It's done this when I've been away on holiday using the 'holiday days left' setting, when there was nothing running except house base load, so it's not that it's unexpected load catching it out.

I normally have it set to look at day 7 for history (but I'm still using the holidays setting as I was away 7 days ago).

gcoan, 2024-02-17:
Hi @iainfogg I was thinking if we could see maybe an earlier plan but more importantly the plan later in the day as predbat runs the battery out prematurely to try to understand why this is happening.

Easiest way to get soc history is the history page in HA and put the battery soc entity in to see it decreasing around the end of the day.

Want to see what level the battery gets down to and how this compares to your min soc and soc keep

springfall2008, 2024-02-18:
Okay I have an idea, the rule that the keep threshold isn't applied fully during the next 4 hours is to prevent force charging the battery at a high rate now rather than waiting.

On the other hand discharging past the keep threshold isn't desirabe.

So my thought is to disable the 4 hour rule once a force discharge starts.

Can you try the version on 'main' and see if it helps?

iainfogg, 2024-02-18:
Thanks @springfall2008 I've just updated to `main` now, I'll see how it goes tonight

iainfogg, 2024-02-18:
Yep, looks like it's sorted it @springfall2008 thanks a lot. Normally at this time (23:10) it would be discharging down towards being empty, but now it's idle having discharged down towards the best SoC Keep setting.

Essentially, it's doing what I expected / hoped it would do - thanks!

iainfogg, 2024-02-19:
Sorted, thanks a lot @springfall2008 - all completed fine last night

iainfogg, 2024-02-26:
Closing as working fine now after the fix

closed 2024-02-26 by iainfogg