springfall2008/batpred #756: Battery not planning to charge

gmarchment, 2024-02-14:
Hi there,

Batpred has been working perfectly until yesterday - there were quite a few updates which I triggered (HA Core, some of the supporting integrations) and Batpred is set to automatically update.

Last night the battery didn't set to charge, and it's not looking like it will plan to charge this evening either.  Any ideas?

[image]

[image]

springfall2008, 2024-02-14:
The error in your status is of concern, is your apps.yaml correct and you don't have duplicate versions?

Also is predbat mode set to control charge?

gmarchment, 2024-02-14:
> The error in your status is of concern, is your apps.yaml correct and you don't have duplicate versions?
> 
> Also is predbat mode set to control charge?

It's weird, as the apps.yaml is definitely right, and there's only one version.

gmarchment, 2024-02-14:
Aha - for some reason Mode had been set to Monitor rather than Control Charge & Discharge.  Thanks Trefor.

closed 2024-02-14 by gmarchment