springfall2008/batpred #759: HACS reports incorrect predbat version

KeithB, 2024-02-14:
**Describe the bug**
Since the ability to update by the updates selector in HA, HACS has become confused and reports the wrong version. First couple of updates on the 7.14.x stream seemed OK but since 7.14.22 HACS does not pick up that the new version is running.

Updates via the selector are successful and confirmed directly from appdaemon logs and checking the predbat.py script directly.

**Expected behavior**
HACS correctly reports the installed predbat version

**Predbat version**
Current version is v7.15.21. HACS reports 7.14.22.

**Environment details**
- HA is a Docker install with appdaemon running as a separate container.
- predbat is a custom appdaemon repository in HACS
- Current predbat install is v7.15.21(first install September/October '23 and then updated regularly since then - no clean re-install done)

**Screenshots**
[image]
[image]

**Log file**
Not sure which logs would be most useful here. Probably don't have the logs available from when it first started but can get logs from todays update to v7.15.21.

gcoan, 2024-02-14:
I have exactly the same problem with a HAOS install and HACS running as an add-on.  It doesn’t seem to recognise that Predbat has been updated - giving me the message “You are running version 7.15.4, version 7.15.21 is available”

image.png (view on web)
image.png (view on web)

gcoan, 2024-02-16:
I have exactly the same problem with a HAOS install and HACS running as an add-on.  It doesn’t seem to recognise that Predbat has been updated

springfall2008, 2024-11-16:
HACS is now depricated

closed 2024-11-16 by springfall2008