springfall2008/batpred #760: Set Charge Low Power Mode

BondBug, 2024-02-16:
**Original Bug Report no 565**
Single AC3 Inverter Economy 7 setup no export.
When low power mode is selected it operates correctly but is sending repeated commands to the Inverter as shown in the GivTCP Log below. There are more off the same above the time line shown below when charging started at 00:30. The inverter charge is being correctly set it just sends the commands multiple times

**New report back as requested below**

Hi Trefor just checked the fix as requested and unfortunately the original problem is still occurring within the GIVTCP Log as below but also there is a minor error message in the Predbat Log when the process is trying to set the charge rate and it returns a message ( WARN: Inverter 0 set charge rate 500 via REST failed got 380 ) even though the GIVTCP Log is showing a change to 500 has been applied. This happened throughout the charge period.

as below

Predbat Log

2024-02-16 00:35:03.627599 INFO pred_bat: Inverter 0 current charge rate is 380 and new target is 500
2024-02-16 00:35:47.146204 INFO pred_bat: WARN: Inverter 0 set charge rate 500 via REST failed got 380
2024-02-16 00:35:47.168867 INFO pred_bat: Configuring charge window now (now 02-16 00:35:00 target set_window_minutes 30 charge start time 02-16 00:30:00
2024-02-16 00:35:47.170257 INFO pred_bat: Inverter 0 Current Target SOC is 80 already at target
2024-02-16 00:35:47.171491 INFO pred_bat: Adjust reserve to default as SOC 73 % is above target 80 % or charging active
2024-02-16 00:35:47.172757 INFO pred_bat: Inverter 0 Current reserve is 4.0 already at target
2024-02-16 00:35:47.180112 INFO pred_bat: Will not recompute the plan, it is 5.0 minutes old and max age is 30 minutes
2024-02-16 00:35:47.180981 INFO pred_bat: Completed run status Charging with Errors reported (check log)

And as before in the GICTCP Log I get repeated commands being applied every 5 mins even though the plan repeat is 30 mins see below this is repeated all through the charge window period not sure if this is meant to happen.

GIVTCP Log same thing happening as before
2024-02-16 00:30:11,195 - write - [INFO] - Setting battery charge rate to: 500 (2)
2024-02-16 00:30:16,825 - write - [INFO] - Setting battery charge rate 2 was a success
2024-02-16 00:30:23,264 - write - [INFO] - Setting battery charge rate to: 500 (2)
2024-02-16 00:30:24,603 - write - [INFO] - Setting battery charge rate 2 was a success
2024-02-16 00:30:32,565 - write - [INFO] - Setting battery charge rate 2 was a success
2024-02-16 00:30:44,714 - write - [INFO] - Setting battery charge rate 2 was a success
2024-02-16 00:30:58,155 - write - [INFO] - Setting battery charge rate 2 was a success
2024-02-16 00:31:06,284 - write - [INFO] - Setting battery charge rate 2 was a success
2024-02-16 00:31:12,874 - write - [INFO] - Setting battery charge rate to: 500 (2)
2024-02-16 00:31:18,473 - write - [INFO] - Setting battery charge rate 2 was a success
2024-02-16 00:31:24,904 - write - [INFO] - Setting battery charge rate to: 500 (2)
2024-02-16 00:31:26,254 - write - [INFO] - Setting battery charge rate 2 was a success
2024-02-16 00:31:34,244 - write - [INFO] - Setting battery charge rate 2 was a success
2024-02-16 00:35:04,981 - write - [INFO] - Setting battery charge rate 2 was a success
2024-02-16 00:35:12,918 - write - [INFO] - Setting battery charge rate 2 was a success
2024-02-16 00:35:20,710 - write - [INFO] - Setting battery charge rate 2 was a success
2024-02-16 00:35:27,308 - write - [INFO] - Setting battery charge rate to: 500 (2)
2024-02-16 00:35:32,713 - write - [INFO] - Setting battery charge rate 2 was a success
2024-02-16 00:35:40,619 - write - [INFO] - Setting battery charge rate 2 was a success
2024-02-16 00:40:04,009 - write - [INFO] - Setting battery charge rate to: 500 (2)
2024-02-16 00:40:08,123 - write - [INFO] - Setting battery charge rate 2 was a success
2024-02-16 00:40:15,883 - write - [INFO] - Setting battery charge rate 2 was a success
2024-02-16 00:40:25,577 - write - [INFO] - Setting battery charge rate 2 was a success
2024-02-16 00:40:37,563 - write - [INFO] - Setting battery charge rate 2 was a success
2024-02-16 00:40:45,263 - write - [INFO] - Setting battery charge rate 2 was a success

Regards

Dave Evans

------ Original Message ------
From "Trefor Southwell" <[notifications@github.com](mailto:notifications@github.com)>
To "springfall2008/batpred" <[batpred@noreply.github.com](mailto:batpred@noreply.github.com)>
Cc "BondBug" <[dpevanslightingdesign@gmail.com](mailto:dpevanslightingdesign@gmail.com)>; "Author" <[author@noreply.github.com](mailto:author@noreply.github.com)>
Date 10/02/2024 19:22:14
Subject Re: [springfall2008/batpred] Set Charge Low Power Mode (Issue https://github.com/springfall2008/batpred/issues/565)

Closed https://github.com/springfall2008/batpred/issues/565 as completed.

springfall2008, 2024-02-16:
I've added a fix on main

BondBug, 2024-02-16:
> From "Trefor Southwell" ***@***.***>
> Cc "BondBug" ***@***.***>; "Author"
> Date 16/02/2024 18:54:13

Great thank you will test as soon as update is available

Regards

Dave Evans

BondBug, 2024-02-17:
Hi Trefor should I still be installing Main or 7.16 to test Low Power 
Mode fix

Regards

Dave Evans

BondBug, 2024-02-18:
Hi Trefor  I have installed Main 7.16.1 and I am pleased to say the 
issues with using Set Charge Low Power Mode is fixed with no multiple 
repeats in the Givtcp Logs and no errors messages in the Predbat  
Appdaemon Logs so all good brilliant work. Does the 7.16.0 Core update 
also have this fix in it as HACS is showing this as a update.

Regards

Dave Evans

BondBug, 2024-02-18:
Hi Trefor just for your info I have also test 7.16.0  to test Charge Low 
Power Mode and that appears to be working as well so all good thanks 
again for all your hard work

Regards

Dave Evans

From "Dave Evans" ***@***.***>

Date 17/02/2024 21:28:03
Subject Re[2]: [springfall2008/batpred] Set Charge Low Power Mode (Issue 
#760)

springfall2008, 2024-02-18:
Great, yes I released this fix in 7.16.0 :)

springfall2008, 2024-11-16:
I think this was fixed, please re-open if not.

closed 2024-11-16 by springfall2008