springfall2008/batpred #766: Predbat crash

SwiftRR, 2024-02-17:
**Describe the bug**
I checked Predbat plan this morning and it had crashed just after 4 am with an 'Exception Error'. I restarted HA and thankfully Predbat came back to life (took about 5 mins). 
Before restarting, I checked other HA installations (such as switching on lights) and they were all working.

I have noticed that all my settings have been switched back to default and I am now changing them to my customised settings (assuming that I can remember them all). I had lost Expert mode which is now back, allowing to access many of my settings.

I checked the AppDaemon Predbat.log(s) and searched for 'error'. I found 72 errors starting at 4:12 am this morning. I have downloaded the file.

I then looked at predbat_debug.yaml. I found a very long file at 4 am but have no idea of what it might tell me.

I have attached both files.

I checked my GE dashboard and my system seems to be continuing with what was its plan last night, basically charging!

**Expected behavior**
No crash!

**Predbat version** 7.15.21, with  automatic update switch enabled.

**Environment details**

- Inverter and battery setup
Raspberry Pi 4 connected to SSD (512 Gb Samsung T7). Very quick response.
Before restarting, other HA installations worked just fine.
Inverter is AC coupled 3.0 (GE) connected to two 8.2 kWh GE batteries.
PV (9.75 kWp with optimisers) interfaced to SolarEdge SE5000H.
I access Predbat via HA on various devices and Predbat hasn't missed a beat since I installed on 30th Jan 2024.

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.
N/A

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)
I have appended .txt to my download as Github didn't allow the raw log file to be uploaded.
[predbat.log.1.txt](https://github.com/springfall2008/batpred/files/14318503/predbat.log.1.txt)

springfall2008, 2024-02-17:
I'm seeing this error which relates to the communication between AppDaemon and home assistant:

2024-02-17 07:05:02.698121 INFO pred_bat: Error: Unable to fetch history for sensor.myenergi_zappi_20464051_charge_added_session
2024-02-17 07:05:02.727637 INFO pred_bat: ERROR: Exception raised 

Did you maybe have something else going on beforehanding, was the machine busy or suchlike?

SwiftRR, 2024-02-17:
Trefor,

Thanks for the timely response.

I have an EV but low mileage and don’t need to charge it often. Last charged on 5th Feb between 3:00 and 3:50 am, 12 days ago (time for HA history?)
The schedule was still there in the myenergi app and I have just deleted it.

I went live with PredBat on 30th January and I did play around with the car charging settings. But my car charging is very ad hoc and I decided that I was best off charging manually so I set num_cars in apps.yaml to 0 (from 1).

I have attached my apps.yaml file.

Should I delete the myenergi HA integration and reinstall it?
Or even not install the integration if I am not using it at the moment.
I do plan to reinvestigate car charging via PredBat at a later date but I am not using it currently.

Should I set any car charging switches to not enabled?
I don’t want this interfering with PredBat which has worked so smoothly for me over the last 2-3 weeks. A great piece of software.

Rob

Rob Ritchie
***@***.***
----------------------------

￼

SwiftRR, 2024-02-17:
Trefor,

In my settings, car_charging_hold is enabled.

With no charging in the last 12 days and hitting the 12 day HA history, could it be that PredBat expected to see an entry and that this caused the exception?
I have disabled the switch for now.
From my view, all the car charging switches are now disabled.

Thanks 

Rob

Rob Ritchie
***@***.***
----------------------------

springfall2008, 2024-02-17:
The sensor is used to subtract previous car charging data from your load data otherwise it will mess up your predictions. I would keep it then and leave car charging hold enable. I think the problem was just down to a HA issue which the restart resolved.

closed 2024-03-03 by springfall2008