springfall2008/batpred #768: Plan length too long

dochtie, 2024-02-17:
**Describe the bug**
I have my plan set to 32 hours but the HTML plan produced is showing 37 hours. There is no black unplanned section on my home battery prediction graph either as there was before.

**Expected behavior**
Prior to 7.16.0 the HTML plan showed me 32 hours. Predbat should only plan the number of hours stated in input_number.predbat_forecast_plan_hours'

**Predbat version**

7.16.0

**Environment details**

- Standard HAOS installer

**Screenshots**
if needed will provide 

**Log file**
If needed will provide.

springfall2008, 2024-02-17:
The number you set in predbat_forecast_plan_hours is the minimum length of the plan, in fact its the time between the first charge and the plan end.

The plan length in apps.yaml sets the maximum plan length, usually 48 hours.

I'd recommend changing your input_number.predbat_forecast_plan_hours to maybe 24

dochtie, 2024-02-17:
Oh interesting. My plan just looked so much longer than normal. Thank you for clarifying.

closed 2024-02-17 by dochtie