springfall2008/batpred #772: Feature Request - Manual Force Freeze Charge

Jingbin-GitHub, 2024-02-17:
**Is your feature request related to a problem? Please describe.**
With a tariff like Agile, when the import price is low but not low enough to charge your battery, but it is low that you would like the battery to hold the charge and let the grid to meet your home load.

I am not sure if this is part of the automation algorithm. But I would like to have at least a manual overwrite.

**Describe the solution you'd like**
Manual Force Freeze Charge (or some other better names), similar to the force charge/discharge/idle.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

springfall2008, 2024-02-18:
Yes this is part of the normal algorithm, as long as you have 'set charge freeze' turned on.

An override is of course possible, will put it on the todo list.

springfall2008, 2024-11-16:
This is now done

closed 2024-11-16 by springfall2008