springfall2008/batpred #784: Predbat isn't correctly taking account of inverter limits when choosing discharge slots

gcoan, 2024-02-21:
**Describe the bug**
Power up event tomorrow so zero rate import from 9:30-12:30.
During that period I know I will be running the washing machine, dishwasher, tumble dryer etc so load will be higher.  To reflect this I set load_scaling to 3.0 in the import_rates_override

Resultant plan:
[image]

Notice that the 11:30 slot is charging AND discharging.  The same happened today in the power up event, the battery started discharging (to gain export revenue) but my load was much higher, no export could occur and so the battery was just cycling unnecessarily. I had to do a force charge in the slot to stop the discharge activity.

Increased load_scaling to 4.0, the load kWh figures increase, but still the discharge:
[image]

Load scaling of 5.0, still discharging:
[image]

Only when load scaling is set to 6.0 (the scaled-up half hour slot load is now 3.27kWh) does Predbat not plan to discharge:
[image]

In apps.yaml my inverter_limit is set to 5000 as its a 5kW gen 1 hybrid, but this is the AC limit, the max charge/discharge rate isn't set in apps.yaml, Predbat is getting it from the inverter itself.

battery rate max scaling's are set to 1.0.

**Expected behavior**
For a 2.6kWh inverter the max charge or discharge in a 30 minute slot must be 1.3kWh.

If the predicted load is > 1.3kWh there is therefore no point in the battery discharging for export revenue as the battery discharge will be soaked up by the house load.  I would expect Predbat to intelligently not decide to discharge in these circumstances.

This intelligent 'don't discharge' decision is happening but at what looks like the wrong load level.

There should be some leeway to the decision whether to discharge or not as after all this is just a load prediction which may be wrong, but I'd expect the leeway to be something like 50 or 75% (e.g. 1/2 hour load>2.3kWh).  It looks from the above screen shots that Predbat is not considering that the inverter discharge limit is per hour, not per half hour, and then its applying some leeway to the 'not discharge' decision.

**Predbat version**
7.16.1
HAOS
Gen 1 hybrid 5kW

springfall2008, 2025-01-01:
As this ticket is old I'm going to close it, please re-open if the issue is there on latest.

closed 2025-01-01 by springfall2008

gcoan, 2025-01-01:
> As this ticket is old I'm going to close it, please re-open if the issue is there on latest.
> 
> 

I haven't seen any similar issues with discharge during the power up event and am not even using load scaling, but will reopen if need be.

Thanks