springfall2008/batpred #785: Solar Charging and Discharging of Battery

dandwhelan, 2024-02-22:
**Describe the bug**
Overnight Prebat discharges my batteries and keeps the house running off solar during the day. Predbat plan is set to discharge and yet keeps charging the battery up. Then discharges back to 6,5 or 4%. The conversion loss from solar DC to Battey DC charging the battery and battery DC to Live Mains AC is small but unnecessary. Why keep charging and exporting the charged-up power? 

**Expected behaviour**
I would expect the power to be exported to the grid instead of charging and discharging the battery to the grid. 

**Predbat version**

7.16.01

**Environment details**

Gen 2 Giv H5k
HAOS

ChatGPT convo.
Me: I have solar panels, a hybrid inverter, and lithium batteries powering the house. Can I get when converting Solar DC to Battery DC, and then Battery DC to Live House AC

Chatbot: Yes, there are energy losses associated with converting solar DC to battery DC, and then battery DC to live house AC. These losses occur due to the inefficiencies of the various components involved in the conversion process. Here's a breakdown of the losses at each stage:

1. Solar DC to Battery DC:

MPPT Charge Controller: This component regulates the voltage and current from the solar panels to maximize the power transferred to the batteries. However, it typically has an efficiency of around 95-98%. This means that 2-5% of the energy generated by the solar panels is lost as heat.

MPPT Charge Controller
Battery Chemistry: Different battery chemistries have varying conversion efficiencies. For example, lithium-ion batteries have an efficiency of around 95-98%, while lead-acid batteries have an efficiency of around 80-85%.
2. Battery DC to Live House AC:

Inverter: The inverter converts the DC electricity stored in the batteries to AC electricity that can be used by your home appliances. Modern inverters have high efficiencies, typically ranging from 95-98%. However, some energy is still lost as heat during the conversion process.

Inverter
Total Losses:

The total energy loss during the entire conversion process can be calculated by multiplying the individual losses at each stage. For example, if you have an MPPT charge controller with 97% efficiency, lithium-ion batteries with 98% efficiency, and an inverter with 96% efficiency, the total energy loss would be:

1 - (0.97 * 0.98 * 0.96) = 0.054, or approximately 5.4%

Therefore, in this example, you would lose around 5.4% of the energy generated by your solar panels due to conversion inefficiencies.

It's important to note that these are just typical efficiencies, and the actual losses in your system may vary depending on the specific components you are using. You can consult the technical specifications of your equipment to determine its efficiency rating.

Here are some additional factors that can affect the energy losses in your system:

Temperature: The efficiency of some components, such as inverters, can decrease at higher temperatures.
Age of equipment: As equipment ages, its efficiency may decline.
System design: A well-designed system will minimize energy losses by using efficient components and properly sizing the equipment for your needs.
By understanding the energy losses associated with your solar power system, you can make informed decisions about your equipment and system design to maximize your energy production and self-consumption.

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

[image]
[apps (1).txt](https://github.com/springfall2008/batpred/files/14373339/apps.1.txt)
[appdaemon1old.log](https://github.com/springfall2008/batpred/files/14373340/appdaemon1old.log)
[appdaemoncurrent.log](https://github.com/springfall2008/batpred/files/14373342/appdaemoncurrent.log)

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

RobinCu, 2024-02-22:
Can you share more of the Predbat HTML plan and the Home Battery Prediction chart showing the last 8-24 hours?  I think I can see what's happening, but seeing 15:30 onwards might confirm it (I appreciate that 23:00 onwards will be based on Nordpool rates).  
Also, there was low overnight pricing 4p-7p/kWh last night; did the battery charge during those slots and discharge at 15p?  
Is that your concern when you say, "Why keep charging and exporting the charged-up power?" or is it the discharging to 4%  that's the concern (or both)?

gcoan, 2024-02-22:
Predbat does a cost optimisation to run the battery to deliver the lowest cost electricity for you.

In the first part of the plan the battery is currently on 6% and from 11:30-14:30 it is discharging the battery down to empty (4%) and any solar generated will be exported at 15p.

Then from 14:30-15:30 your export rate is set to 5p, so it's not worth exporting and instead any solar is allowed to accumulate in the battery.

From 15:00 with the import rate low (10.18p) Predbat is starting to charge the battery to give you enough stored electricity to see you through the evening peak rates.

I would question the export rate from 14:30-15:30, this looks very low. Is there a price override been set?

The import and export rates, the figure in brackets is the effective rate after losses have been taken into account.
There's an FAQ that explains how the different loss figures are used by Predbat https://springfall2008.github.io/batpred/faq/#the-plan-doesnt-charge-or-discharge-when-i-expect-it-to

Otherwise, which bits of this plan would you want Predbat to do differently?

dandwhelan, 2024-02-22:
Here is my dashboard at this present time.

Red = Solar Generated 
Yellow = House power being used
Pink = Battery charging 
Blue = Export
[image]

Here is the plan for the rest of the day. I removed the override of export between certain times to simplify things. 

[image]

> "Why keep charging and exporting the charged-up power?" or is it the discharging to 4% that's the concern (or both)?

Yes, **"Why keep charging and exporting the charged-up power?"** is my key question. 

On a side note, whenever I change predbat to monitor mode, it charges the battery.

[image]

RobinCu, 2024-02-22:
The plan looks OK to me based on your settings. Right now, it has surplus solar, which (based on the PV and house load forecasts) it believes it can safely export the surplus.  Then, at 15:00, it will charge at a rate after losses of 12.16p.  In an ideal plan, I'd say it should charge a little in the 14:30 slot at 14.52p to cover the 19:30 to 20:30 period, but the amounts involved and the unknown variables mean it's just noise.

The discharge in the 00:00/00:30 slot is exporting energy at 12.68p that it has imported from 22.30 to 00:15 (approx) at an average of 10.5p after losses. Given that Predbat's primary objective is to save/make as much money as possible, it's working as intended.  You can, of course, change it to Control Charge mode, and it will still supply the house load as required without force exporting.

I'm not sure exactly what you mean re: the side note.  That graph just shows cost, which correlates with your house load forecasts. i.e. there is a jump in the cost on the graph at around 19:00, and your plan shows a forecast load of 2.47 kWh in 30 minutes.  Another jump at midnight is the standing charge (you can comment this out in apps.yml if you just want to see unit costs).  Maybe I'm missing something, though.  The Home Battery Prediction chart / Predbat Plan should show no charge/discharge when you change to Monitor mode (after a minute or so for the plan to recalculate).

dandwhelan, 2024-02-22:
OK. I just feel the charging with solar and discharging with the charge on the battery is unnecessary. If you think it's fine I'll close the issue

gcoan, 2024-02-22:
> OK. I just feel the charging with solar and discharging with the charge on the battery is unnecessary. If you think it's fine I'll close the issue 

Like Robin I think your plan is OK.

Your solar export rate from 14:00-15:00 is 12.68p after losses, whereas your import rate from 15:00-15:30 is 12.16p.

So it makes financial sense to export your excesss solar and then import at the slightly lower rate.
The import from 15:00 to 15:30 charges your battery to about 30% and this charge is then released from 16:00 to sometime in the 19:00 slot when your battery runs out. This period being the higher import rates that the charge beforehand avoids you incurring.

If you want to reduce use of the battery a bit when the savings are marginal you can increase the predbat_metric_battery_cycle or increase the predbat_metric_min_improvement (_discharge) - details as per the FAQ I linked earlier

closed 2024-02-23 by dandwhelan