springfall2008/batpred #786: Add support for Solar Assistant

xhemp, 2024-02-22:
Would be great to have support for Solar Assistant ( https://solar-assistant.io/ ). It supports many types of inverters and the beta version has a proper power management that works fine with Growatt (my case) inverters as well.
I saw another issue opened here for Huaweii ( #684 )  so I will try to play a bit with the template yaml and see where I get, just wanted to get this on the radar (and also confirm if this is something doable or not).
Thanks!

springfall2008, 2024-03-03:
Could make sense, but can share details on how to control it in HA?

xhemp, 2024-03-03:
Hey, thanks for getting back to me!
So the way I use SA with Home Assistant via MQTT:
For example, there's a switch called Work mode priority, there you select Load first, Battery first or Grid first. That is tied to the device_mode topic on MQTT.
Attached is a screenshot of all the MQTT topics for Solar Assistant:

[image]

If you need anything else or have any other questions, let me know.

springfall2008, 2024-03-04:
Do these MQTT topics appear as HA entities?

I would need a list of HA entities that hold the basic information required for apps.yaml together with some example automations used to charge and force discharge the battery

xhemp, 2024-03-04:
They do, yes and sure thing, here's the entities list: 
[entities_list.txt](https://github.com/springfall2008/batpred/files/14480640/entities_list.txt)
Please, let me know if you need anything else!

springfall2008, 2024-03-07:
I guess I don't really know what these do without context.

Can you take this template and try to map these sensors to the line items listed here: 

https://github.com/springfall2008/batpred/blob/main/templates/solaredge.yaml

Then if you can create a simple automation in HA to start and stop charging the battery from grid and to discharge the battery to the grid that can go into the configuration also.

xhemp, 2024-03-11:
@springfall2008 Sure thing, I am abroad this week but I will give this a go next week, thanks!

xhemp, 2024-03-17:
Those are the sensor values I came up with ( I added a few questions to understand what's the need on some of them):

  battery_rate_max:
    - number.growatt_spa_battery_first_stop_charge
  charge_rate: ( In percentage works? Mine, in watts, is 3kW/h charge)
    - number.growatt_spa_free_mode_charge_rate
  discharge_rate: ( In percentage works? Mine, in watts, is 3kW/h discharge)
    - number.growatt_spa_free_mode_discharge_rate
  battery_power:
    - sensor.growatt_spa_battery_power
  pv_power:
    - sensor.growatt_spa_pv_power
  soc_max: (is this the maximum in watts? Mine is 6kW, as the charge/discharge rate, if needed, 
            I can create a custom template with those values set)
    - sensor.?

For the automation, here they are:

  # Services to control charging/discharging
    service: select.select_option
    entity_id: "select.growatt_spa_work_mode_priority"
    option: "Battery first"
  charge_stop_service:
    service: select.select_option
    entity_id: "select.growatt_spa_work_mode_priority"
    option: "Load first"
  discharge_start_service:
    service: select.select_option
    entity_id: "select.growatt_spa_work_mode_priority"
    option: "Grid first"

Let me know if this is what you were looking for and if you need any changes or an explanation.

springfall2008, 2024-03-18:
charge_rate - should be a setting in Watts e.g. 3000=3kW.
You can just hard-wire it as 3000 for now
discharge_rate - as charge rate

soc_max is a value in kWh which is the size of the battery e.g. 10 would be a 10 kWh battery

On the services, does Grid first force the battery to discharge at full rate?

xhemp, 2024-03-18:
> charge_rate - should be a setting in Watts e.g. 3000=3kW.
> You can just hard-wire it as 3000 for now
> discharge_rate - as charge rate
>
> soc_max is a value in kWh which is the size of the battery e.g. [...]

Thanks for clarifying.
Grid first forces the battery to discharge at whatever percentage is set on another sensor ( number.growatt_spa_grid_first_discharge_rate ).

xhemp, 2024-11-02:
I would love to get this one sorted, any help is appreciated!

springfall2008, 2024-11-16:
Did you ever get this to work?

xhemp, 2024-11-16:
I didn't, no, but I am keen on trying again

springfall2008, 2024-11-16:
The service API is a bit more comprehensive now e.g. for Solax we configure apps.yaml as

```yaml
  # Service for start/stop
  charge_start_service:
    service: select.select_option
    entity_id: select.solax_manual_mode_select
    option: "Force Charge"
  charge_stop_service:
    service: select.select_option
    entity_id: select.solax_manual_mode_select
    option: "Stop Charge and Discharge"
  discharge_start_service:
    service: select.select_option
    entity_id: select.solax_manual_mode_select
    option: "Force Discharge"
```

xhemp, 2024-11-17:
Thanks for that, I will have a go on this next week and let you know.

xhemp, 2024-11-19:
I tried to change the SX4 template to this:

https://pastebin.com/2Jf07e8F

Got an error for InverterType so not sure on what I need to set it there.
Anything else I need to change, do you think?

xhemp, 2024-11-19:
managed to add SA (Solar Assistant) to the config.py, same as the SX4 and was able to pass the inverter type error.
Looks like it's working just confirming if it's doing what it's supposed to do now, will update you later today.

xhemp, 2024-11-23:
Something I also realised is that the holidays days left are not decreasing at midnight (I'm not at home at the moment).

xhemp, 2024-11-24:
Looks like the errors I am getting above happen when the inverter has to be set to export, when I check Solar Assistant, it doesn't show it exporting so I am unable to export it, unfortunately.

xhemp, 2024-12-06:
@springfall2008 I would really appreciate if you can add the SA config (with all false at the moment) to the config.py for now.

PR #1723 (2024-12-13): Add solar assistant to predbat
Added Solar Assistant to PredBat. 
I believe this will help whoever wants to use PredBat with Solar Assistant and would allow people (like me) to upgrade to new versions without modifying the config.py.
This resolves #786
closed 2024-12-13 by springfall2008