springfall2008/batpred #787: iBoost broken in version 7.16.x

campbellcopeland, 2024-02-22:
**Describe the bug**

iBoost isn’t being calculated on the plan in 7.16.x when using settings for cheaper than gas. Not seeing any planned values but dropping back to 7.15.x does.

**Expected behavior**
When Agile rates are less than gas, iBoost should be planned. 

**Predbat version**

7.16.1

**Environment details**

N/A

**Screenshots**
N/A

**Log file**
Not captured at this point

springfall2008, 2024-03-03:
Fixed on main, will be in the next release

springfall2008, 2024-03-04:
Now released

campbellcopeland, 2024-03-04:
I don’t think this is quite right. Earlier today it planned multiple iBoost sessions when rate was more than gas, irrespective of scaling used. Tonight, it caused an empty plan to appear with no planned charge/discharge.

Will retry & capture logs tomorrow.

springfall2008, 2024-03-07:
Fix now on main, sorry I changed something I shouldn't and it was just boosting on charging regardless. If you can test it that would be great

campbellcopeland, 2024-03-08:
I’m on agile so only planning iBoost to trigger my automation for Tuya immersion control when rates are cheaper than gas - just now, agile hasn’t dropped low enough for slots to be planned. But, it’s definitely looking better than previous release.

7Van737, 2024-03-08:
@campbellcopeland I note above you mention setting the 'iBoost gas price scaling' to 0.8 to equate to 80% boiler efficiency.   There is no current detail of the workings of the scaling factor in the documentation (@gcoan) but I think you need a reciprocal of 0.8, i.e 1.25 to equate to an 80% gas boiler efficiency.   

@springfall2008 Is there a reason the 'iBoost Max Energy' is restricted to 5 kWh?   Pre having my battery I've had many days when I've heated the tank with around 12kWh from excess solar.  I expect less to be available gong forward but are still consuming more than 5kWh through the immersion when Agile rates are low.

gcoan, 2024-03-08:
@campbellcopeland thanks for highlighting this. I'll look into updating the documentation

springfall2008, 2024-03-08:
>Is there a reason the 'iBoost Max Energy' is restricted to 5 kWh?

AFAIK its not you just change the value in HA?

gcoan, 2024-03-08:
> >Is there a reason the 'iBoost Max Energy' is restricted to 5 kWh?
> 
> 
> 
> AFAIK its not you just change the value in HA?

In predbat.py iboost_max_energy seems to be set to have a max value of 5:

    {
        "name": "iboost_max_energy",
        "friendly_name": "IBoost max energy",
        "type": "input_number",
        "min": 0,
        "max": 5,
        "step": 0.1,
        "unit": "kWh",
        "enable": "iboost_enable",
        "default": 3.0,
    },

springfall2008, 2024-03-08:
> In predbat.py iboost_max_energy seems to be set to have a max value of 5:
>
> ```
> {
>     "name": "iboost_max_energy",
>     "friendly_name": "IBoost max energy",
>     "type": "input_number",
>     "min": [...]

I can change that, but AFAIK it doesn't prevent you from entering bigger numbers

gcoan, 2024-03-08:
@campbellcopeland out of interest how are you controlling your iboost from the predbat binary sensor? 
I've heard that there isn't a home assistant integration?

Thanks

campbellcopeland, 2024-03-08:
@gcoan i don’t have an Iboost, just a Tuya enabled switch for my immersion heater that I have an automation to trigger whenever Predbat sets the appropriate flag. So, it’s either on or off at 3kW - only trigger when cheaper than gas.

gcoan, 2024-03-09:
> @gcoan i don’t have an Iboost, just a Tuya enabled switch for my immersion heater that I have an automation to trigger whenever Predbat sets the appropriate flag. So, it’s either on or off at 3kW - only trigger when cheaper than gas. 

Thanks, I plan to make the documentation a bit more generic, covering immersion heaters, EDDI devices etc.
Someone on the givenergy community forum said its not possible to remotely control an iboost so wanted to understand what people are actually doing with the predbat function

springfall2008, 2024-03-09:
> Thanks, I plan to make the documentation a bit more generic, covering immersion heaters, EDDI devices etc.
> Someone on the givenergy community forum said its not possible to remotely control an [...]

Some people have added one of those mechanical button pushers to iBoost

campbellcopeland, 2024-03-09:
I spent under £40 to replace my existing fuse spur with something similar to this. couldn’t justify the cost of a proper solar diverter due to export rates vs Octopus Gas Tracker price differences. Allows the Tuya integration in Home Assistant.

https://amzn.eu/d/3soSIA0

7Van737, 2024-03-09:
I do have an iBoost which has diverted 16MWh to my hot water cyclinder in the 10 years that I’ve owned it.   As Geoffrey says it has no smart features and in order perform a "boost", i.e. operate the immersion at maximum rate, a manual button must be pressed on the front of the unit - each press adds 15 mins to the total boost period - up to 120 mins max.  I did consider a ‘Fingerbot’ but saw some comments saying they eventually fall off.

I have therefore installed two Sonoff POW 20W switches in a din rail enclosure - one to isolate the iBoost and the other to bypass it.   When generating solar, the iBoost is live and the bypass closed.   The bypass is opened whenever Agile prices fall below a set price.  I have all this controlled by an automation which ensures the iboost isolating and bypass switches are never both open at the same time as otherwise the iboost is likely to be damaged.

gcoan, 2024-03-09:
thanks guys for sharing how you've automated your hot water.  Its not strictly within Predbat scope but I'll add a little in to cover different options and how to link predbat to those options

springfall2008, 2024-03-10:
Nice one

I did start to develop Predheat which can feed data into Predbat but I don't know how accurate it is

gcoan, 2024-03-17:
> @campbellcopeland I note above you mention setting the 'iBoost gas price scaling' to 0.8 to equate to 80% boiler efficiency. There is no current detail of the workings of the scaling factor in the [...]

thanks, I have updated and expanded the 'iboost' documentation to be more generic about different types of solar diverters, not just iboost and added a generic 'starter' automation

Having checked the code, @campbellcopeland yes predbat_iboost_gas_scaling does need to be set to the reciprocal of the gas boiler efficiency as the value entered is multiplied by the gas price rate.

Updated docs here:
https://github.com/gcoan/batpred/blob/main/docs/customisation.md

springfall2008, 2024-11-16:
I think this was fixed, please re-open if not.

closed 2024-11-16 by springfall2008