springfall2008/batpred #788: iBoost Manual Switch

dandwhelan, 2024-02-23:
**Is your feature request related to a problem? Please describe.**
I have a on-off switch in HA charging up my hot water tank as a Solar PVD "iBoost". I don't have an iBoost, but there is a feature I'm interested in using, "switch.predbat_iboost_gas'" iBoost when cheaper than gas. I like the ability to specify a manual switch to trigger as my iBoost. 

**Describe the solution you'd like**

I would like an option to specify a switch to trigger all the predbat iBoost functions for my non-iBoost. I don't know if this is possible. This would be helpful to do it via predbat to make sure the battery is charging when it's running.  ('switch.predbat_iboost_charging') IBoost when battery charging 

**Describe alternatives you've considered**
At the moment, I've automation on HA.

springfall2008, 2024-03-03:
You want Predbat to trigger a switch when iboost is predicted to run?

There is already a sensor for this:

binary_sensor.predbat_iboost_running

You just need to link it via an automation to the devices you want to control, does that not work?

dandwhelan, 2024-03-03:
Yeah I want predbat to trigger an on and off switch for iboost. The on and
off as a manual entity.

dandwhelan, 2024-03-03:
I'll check it out. Thanks

springfall2008, 2024-03-03:
One thing is iBoost appears to be broken, but I've made a fix on main that I will release soon once it's had more testing on the other changes

gcoan, 2024-03-17:
@dandwhelan I have been updating the predbat documentation for iboost, and to confirm, the entity you require to trigger an action when iboost is planned to run is binary_sensor.predbat_iboost_active

(not _running) as above

updated docs here: https://github.com/gcoan/batpred/blob/main/docs/customisation.md

dandwhelan, 2024-04-14:
This was fixed and with @springfall2008 suggestions I managed to create an automation.

closed 2024-04-14 by dandwhelan