springfall2008/batpred #790: Monitor mode charging battery via grid

dandwhelan, 2024-02-23:
**Describe the bug**
Switching to monitor mode charges the battery

**Expected behavior**
Monitor the system 

**Predbat version**

7.16. 1

**Environment details**

- gen 2 h5 
- Standard HAOS installer 
- Anything else? It's been like this for about a week. 

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.
[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)
[appss.txt](https://github.com/springfall2008/batpred/files/14390814/appss.txt)
[appdaemnon.txt](https://github.com/springfall2008/batpred/files/14390815/appdaemnon.txt)

dandwhelan, 2024-02-23:
[image]

gcoan, 2024-02-24:
> To be clear, in Monitor mode, the plan will show you what Predbat would like to do to optimise charging/discharging but can’t (and won’t) because Control Charge & Discharge is not active.

That bit is not quite correct Robin.

In Monitor mode, Predbat does not plan any charge or discharge activity, it only monitors your current inverter settings, house load and predicted solar charging and shows what would happen in the future.

The 'what would predbat like to do but can't' is achieved by setting mode to Control Charge & Discharge (so Predbat calculates an optimal plan) and read-only to On (so Predbat can't enact that plan and make any inverter changes)

RobinCu, 2024-02-24:
@gcoan you’re quite right. This is why I shouldn’t answer emails before 10am!

I think therefore it might just be a timing issue that Monitor mode was enable at 21:56, then Control & Discharge was selected at 21:58, and the effect wasn’t seen in the plan.

 I can double check the rest of the log in a bit but I stopped when I saw these two events.

dandwhelan, 2024-02-24:
> What is it that indicates the battery is charging? If it something in the GE app or GE Portal?
> 
> If I check the log at 2024-02-23 21:56:15.061274 INFO pred_bat: select_event: mode, select.predbat_mode = Monitor 2024-02-23 21:56:15.067578 INFO pred_bat: Set reset inverter true due to reset_inverter_force on item mode 2024-02-23 21:56:15.072716 INFO pred_bat: Set reset inverter force true due to reset_inverter_force on item mode
> 
> That all looks fine.
> 
> At the end of that recalculation it says: 2024-02-23 21:56:40.242281 INFO pred_bat: Completed run status Idle
> 
> 2 minutes later the control method was returned to Control Charge & Discharge.
> 
> The plan it shows on the Predbat Plan (your first screenshot), looks to be OK.
> 
> To be clear, in Monitor mode, the plan will show you what Predbat would like to do to optimise charging/discharging but can’t (and won’t) because Control Charge & Discharge is not active.

That was more than likely me mess with the settings. You want to be checking around 23:00.

> 2024-02-23 23:05:42.989575 INFO pred_bat: Base charge    window [ 23-02 22:30:00 - 24-02 08:00:00 @ 0p 100%, 24-02 22:30:00 - 25-02 08:00:00 @ 0p 100%, 25-02 22:30:00 - 26-02 08:00:00 @ 0p 100% ]
2024-02-23 23:05:42.994312 INFO pred_bat: Base discharge window [ 23-02 05:00:00 - 23-02 05:31:00 @ 0p 100.0%, 24-02 05:00:00 - 24-02 05:31:00 @ 0p 100.0% ]
2024-02-23 23:05:42.998974 INFO pred_bat: Reset inverter settings to safe mode (set_charge_window=True set_discharge_window=False force=mode)
2024-02-23 23:05:43.004775 INFO pred_bat: Inverter 0 Current Target SOC is 100 already at target
2024-02-23 23:05:43.010175 INFO pred_bat: Inverter 0 Current reserve is 4.0 already at target
2024-02-23 23:05:43.017890 INFO pred_bat: Inverter 0 Adjust force discharge to False, change times from 05:00:00 - 05:31:00 to None - None
2024-02-23 23:05:43.023422 INFO pred_bat: Include original charge start 02-23 22:30:00, keeping this instead of new start 02-23 23:00:00
2024-02-23 23:05:43.027358 INFO pred_bat: Charge window will be: 2024-02-23 22:30:00+00:00 - 2024-02-24 08:00:00+00:00 - current soc 97 target 100
2024-02-23 23:05:43.032822 INFO pred_bat: Configuring charge window now (now 02-23 23:05:00 target set_window_minutes 30 charge start time 02-23 22:30:00
2024-02-23 23:05:43.042910 INFO pred_bat: Inverter 0 Current Target SOC is 100 already at target
**2024-02-23 23:05:43.048649 INFO pred_bat: Adjust reserve to default as SOC 97 % is above target 100 % or charging active**
2024-02-23 23:05:43.055516 INFO pred_bat: Inverter 0 Current reserve is 4.0 already at target
2024-02-23 23:05:43.105426 INFO pred_bat: IBoost model today updated to 0.0
2024-02-23 23:05:43.109274 INFO pred_bat: Car charging Manual SOC current is 61.46399999999998 next is 61.46399999999998
**2024-02-23 23:05:43.112706 INFO pred_bat: Completed run status Charging**
2024-02-23 23:05:43.175379 INFO pred_bat: Creating predbat dashboard at /homeassistant/predbat_dashboard.yaml
2024-02-23 23:05:43.273059 INFO pred_bat: --------------- PredBat - update at 2024-02-23 23:05:43.270456+00:00 with clock skew 0 minutes, minutes now 1385
2024-02-23 23:05:43.302200 INFO pred_bat: Using cached GITHub data for https://api.github.com/repos/springfall2008/batpred/releases age 50.7 minutes
2024-02-23 23:05:43.305985 INFO pred_bat: Predbat /homeassistant/appdaemon/apps/batpred/predbat.py version v7.16.1 currently running, latest version is v7.16.1 latest beta v7.16.1
2024-02-23 23:05:43.309813 INFO pred_bat: Autoupdate: Currently up to date
2024-02-23 23:05:43.387161 INFO pred_bat: Inverter type GE forecast_hours 48 max_windows 96 num_cars 1 debug enable is False calculate_plan_every 10
2024-02-23 23:05:43.417350 INFO pred_bat: Cars 1 charging from battery False planned [False], charging_now [False] smart [True], plan_time ['07:00:00'], battery size [61.7], limit [61.7], rate [7.2]
**2024-02-23 23:05:43.421822 INFO pred_bat: Predbat mode is set to Monitor**
2024-02-23 23:05:43.592295 INFO pred_bat: IBoost energy today from sensor reads 0.0 kWh
2024-02-23 23:05:45.467988 INFO pred_bat: Found 11521 load_today datapoints going back 8 days
2024-02-23 23:05:48.619167 INFO pred_bat: Current data so far today: load 16.51 kWh import 18.9 kWh export 5.4 kWh pv 8.4 kWh
2024-02-23 23:05:48.623389 INFO pred_bat: Fetching futurerate data from https://www.nordpoolgroup.com/api/marketdata/page/325?currency=GBP
2024-02-23 23:05:48.629170 INFO pred_bat: Return cached futurerate data for https://www.nordpoolgroup.com/api/marketdata/page/325?currency=GBP age 424.2 minutes
2024-02-23 23:05:48.634170 INFO pred_bat: Future rates - peak rate is 960 - 1140 minutes premium import 14 export 0
2024-02-23 23:05:48.713195 INFO pred_bat: Loaded 192 datapoints of futurerate analysis

Maybe once changing form 'Charge and Discharge' to "Monitor" while in Charging mode it isnt resetting back to Eco run off battery? It's just strange how it forward plans a schedule with "Charging" sessions.

dandwhelan, 2024-02-24:
[appdaemon.txt](https://github.com/springfall2008/batpred/files/14392359/appdaemon.txt)

Logs from overnight. I've replicated the bug this morning when the Invertor was charging the battery from the grid. I switched it from 'Charge and Discharge' to 'Monitor' and it kept the plan / kept charging the battery from the grid.

RobinCu, 2024-02-25:
> [appdaemon.txt](https://github.com/springfall2008/batpred/files/14392359/appdaemon.txt)
>
> Logs from overnight. I've replicated the bug this morning when the Invertor was charging the battery from the [...]

Apologies, I've only just had a chance to look at this, but if I'm looking at the right place - you replicated this at 
09:08:34 switched to Monitor. 
09:08:53 Reset inverter settings to safe mode (set_charge_window=False set_discharge_window=False force=mode)
09:08:54 run status was Idle. 

09:12:49 switched to Control charge & discharge
09:14:01 Completed run status Hold charging

09:35:37 switched to Monitor mid-calculation.  The calculation was completed at 09:36:47 as Charging, but at 09:36:48 started the recalculation in Monitor and completed at 09:37:03 as Completed run status Idle.

Based on what I can see, it is working as expected, but it is a timing issue, amplified slightly when the change is mid-calculation.

Let me know if I've been looking at the correct point of the log.

dandwhelan, 2024-02-25:
Sorry I don't agree. I'm just reporting what I'm seeing. Maybe try it on
yours?

It keeps charging the battery once it goes into monitoring mode, and even
keeps a charging plan for it.

This only happens if the previous mode plan is, at the time of changing to
monitor, charging the battery. While it's charging up, set it to monitoring
mode.

springfall2008, 2024-03-03:
Fix made on main for the next release

dandwhelan, 2024-04-14:
This was fixed as stated above. Thanks as always

closed 2024-04-14 by dandwhelan